JMRI/JMRI #12975: Roster function labels - LOCK indicator (for LATCHing functions)

philguk, 2024-03-22:
The recently added JMRI throttle labels added a checkbox called "LOCK"

This controls whether a function is LATCHING or NON LATCHING

Most DCC systems (and therefore users) do refer to the action of functions as "latching" or "momentary" so labelling this option "LOCK" is not intuitive (and has already lead to questions being asked on the JMRI mailing list)

Could we please change this option to read LATCH instead of LOCK?

Thanks

danielb987, 2024-03-22:
Background:
https://groups.io/g/jmriusers/topic/105087519

Lionel642, 2024-03-23:
Usually, I handle the throttle stuff, but for this one, being a french speaker, I'll let you, English and American users decide. If we go for this change, we have to make is consistent, in the throttle function button properties screen of a throttle window and in the documentation. I can do it, but I can't decide if this is right.

danielb987, 2024-03-23:
I have asked about this on the JMRI developers list.
https://jmri-developers.groups.io/g/jmri/message/9936

philguk, 2024-03-23:
I can’t figure out how to add my 2c on that message, but I am incredibly concerned about INVERTING the meaning of the option (from “lock/latch” to “momentary”!)

Phil G

https://jmri-developers.groups.io/g/jmri/message/9936

closed 2024-04-09 by danielb987