JMRI/JMRI #12979: Decoderpro Roster: update links for help files

tomgmyrick, 2024-03-24:
Hello,

The window (context specific) help files for the "Function Labels" and "Roster Media" tabs under the Roster button "Labels and Media" are too general to decoderpro, and should bring up the specific help files for these tabs.  I needed to ask for help on the JMRI user group to understand how these tabs work.  The help files that should come up are as follows:

- Function Labels:  https://www.jmri.org/help/en/manual/DecoderPro3/Adv_FunctionLabel.shtml
- Roster Media: https://www.jmri.org/help/en/html/tools/throttle/RostersMediaPane.shtml

tomgmyrick, 2024-05-13:
Commenting on this issue because it has become stale.

bobjacobsen, 2024-08-12:
See PR #13323

closed 2024-08-21 by bobjacobsen