JMRI/JMRI #12980: Exact mode turnout changes state without feedback

n3ix, 2024-03-24:
Here are two examples of the current behavior of the turnout state when Exact Feedback is specified for the turnout.

Case 1 correct operation: Loconet Turnout set for Exact Mode Feedback.   Current state is Closed.  Using Turnout Tool send command to set a turnout to Thrown.  JMRI shows the state as Inconsistent as expected.  Once the proper feedback command is received from a stationary decoder the Turnout state is shown as Thrown as expected.

Case 2 undesired operation: Loconet Turnout set for Exact Mode Feedback.   Current state is Closed.  Using Turnout Tool send command to set a turnout to Thrown.  JMRI shows the state as Inconsistent as expected.  Send another command to set the same turnout to Thrown. JMRI immediately shows the state as Thrown, even though no exact feedback has been received on Loconet.

I would expect the state in Case 2 to remain Inconsistent unless an exact feedback message is received on Loconet that indicates the turnout is now Thrown?

bobjacobsen, 2024-03-24:
A LocoNet trace of this interaction would be really helpful. In case 2, has the exact feedback message that says "no longer thrown" come in when this happens?

n3ix, 2024-03-25:
@bobjacobsen FYI traces below.  (fyi a user on the jmri list has also reported this issue.)

Case 1
08:32:27.600: [B0 37 19 61]  Requesting Switch at LT1208 to Thrown (Output On).
(LT1208 shown as Inconsistent in JMRI at this point)
08:32:27.607: [B1 37 69 10]  Turnout LT1208 () Switch input is Thrown (input on).
08:32:27.699: [B0 37 09 71]  Requesting Switch at LT1208 to Thrown (Output Off).
08:32:30.795: [B1 37 59 20]  Turnout LT1208 () Aux input is Closed (input off).
(LT1208 shown as Thrown in JMRI at this point)

Case 2
08:33:47.295: [B0 37 19 61]  Requesting Switch at LT1208 to Thrown (Output On).
08:33:47.395: [B0 37 09 71]  Requesting Switch at LT1208 to Thrown (Output Off).
(LT1208 shown as Inconsistent in JMRI at this point)
08:33:50.349: [B0 37 19 61]  Requesting Switch at LT1208 to Thrown (Output On).
08:33:50.450: [B0 37 09 71]  Requesting Switch at LT1208 to Thrown (Output Off).
(LT1208 shown as Thrown in JMRI at this point)

bobjacobsen, 2024-03-25:
@n3ix Thank you for the traces.  

In case 1, the turnout starts to move quickly, saying that the turnout is in motion at 27.607. The motion completes at 30.795.  So the Thrown here is correct.

 In case 2, the turnout hasn't moved even though 3 seconds have elapsed.  The exact feedback is still indicating Closed, so the 2nd Thrown command shouldn't be changing the turnout. 

I'll look into this.

n3ix, 2024-03-25:
@bobjacobsen Thank you Bob

n3ix, 2024-03-25:
@bobjacobsen I manually sent the turnout command twice by clicking the Turnout Tool THROWN button.  Here is the log if I only click the button once:

08:55:43.708: [B0 37 19 61]  Requesting Switch at LT1208 to Thrown (Output On).
08:55:43.808: [B0 37 09 71]  Requesting Switch at LT1208 to Thrown (Output Off).
LT1208 shown as Inconsistent

n3ix, 2024-03-25:
@bobjacobsen the issue only occurs when I click the THROWN button a second time

n3ix, 2024-03-25:
@bobjacobsen based on your comment about the 3 second timer here is what I think might be an explanation.  As you know with Exact feedback there should be 2 messages: the first one when the turnout starts to move and the second when it completes the move.   If JMRI sees _no_ messages in 3 seconds, it concludes that the action state of the turnout was already the commanded state.   That seems ok, it just isn't working that way.
1) if there is no feedback received, the state remains Inconsistent until another command is sent to that turnout
2) upon the next command to that turnout, whether THROWN or CLOSED, the state changes to match the command

bobjacobsen, 2024-03-25:
Could you check if it shows the same problem when going to CLOSED please?

n3ix, 2024-03-25:
@bobjacobsen yes exactly the same behavior with CLOSED.  This has all been done using the Turnout Tool and the state it displays.  With no feedback, it shows "Inconsistent" when clicking THROWN if the current displayed state is CLOSED, or when clicking CLOSED if the current displayed state is THROWN.  After one button click the displayed state remains Inconsistent until clicking again on the same button.  At that point the displayed state matches the button.

I just tried some tests using the Turnout Table.  The "Feedback" column always reflects the latest feedback message received from the decoder (if any).  The displayed state is only shown as Inconsistent from the point an exact feedback message indicates that the turnout is in motion until another exact feedback message indicates the turnout has reached the Closed or Thrown position.

I also added the turnout to a Layout Editor panel.   The icon displayed matches the "Feedback" column in the Turnout Table, so it is based solely on the exact feedback messages from the device.

In no case did anything change after a delay.

danielb987, 2024-03-25:
@Pugwash1 
@devel-bobm 
Just a note about this thread if it's something that might interest you. See in particular this comment:
https://github.com/JMRI/JMRI/issues/12980#issuecomment-2018338220

devel-bobm, 2024-03-26:
I was wondering about "exact feedback" and "a 3 second timeout" when I saw the previous messages.  But I did not have anything to add - too many people have too many ideas about how LocoNet "ought" to work.  

Now that I've been sucked in, here are some things relating to this issue:

- Why 3 seconds?  Why exactly 3 seconds?  Why not adjustable?  Why not adjustable by particular turnout?  Why does _any_ turnout need to be "re-thrown" (from the LocoNet message perspective)?  [FYI:  I have some turnouts (C/MRI, not LocoNet!) which take 5 to 6 seconds to throw...]

- Is this "3 second rule" related to "command station turnout request buffer overload"?  Or something else?

- Is the particular user problem related to a DCC-controlled switch machine or a LocoNet-controlled one?  Probably one that is LocoNet-controlled, given @n3ix 's involvement.  But it could be a DCC-track controlled switch, in which case the challemge is possibly very different!

- The code line:
  `if (getFeedbackMode()==MONITORING && !on || getFeedbackMode()==MONITORING && on && !_useOffSwReqAsConfirmation || getFeedbackMode()==DIRECT) {`
    from `LnThrottle` is, at best, awkward.  I hope someone who understands Java priority of operations more than I has looked carefully at this.
 
- Perhaps I am interpreting the use of "on" and "off" incorrectly, thinking that it can relate to "on messages" and "off messages".  With that interpretation: will LocoNet "exact feedback" with <OPC_SW_REP> (0xB1) messages work for a turnout when the user has _disabled_ "switch off" messages?  Or with some recent throttles (and/or throttle firmware revisions) that do not send "switch off" messages?  The `LnTurnout` code leads me to believe at least part of the logic will not work.  But will it give appropriate functionality with just "on" messages?  I have not investigated...

- From Robin's loconet logs, there is a basic problem depending on _when_ the LocoNet messages show up in JMRI's domain.  Some implementations can show _more than one_ LocoNet message "arriving" _at the same time_.  Presumably, those implementations have more trouble with "case 2" than others.  What LocoNet connection type has this problem been seen on?  Perhaps with a Digitrax CS with integrated USB (which can have other problems) or PR4?

Enough "thoughts" for now...

danielb987, 2024-03-26:
> The code line:
> if (getFeedbackMode()==MONITORING && !on || getFeedbackMode()==MONITORING && on && !_useOffSwReqAsConfirmation || getFeedbackMode()==DIRECT) {
> from LnThrottle is, at best, awkward. I hope someone who understands Java priority of operations more than I has looked carefully at this.

&& has higher precedence than || so the items that has && will be calculated before || . So the expression looks right to me.

https://docs.oracle.com/javase/tutorial/java/nutsandbolts/operators.html

n3ix, 2024-03-26:
@devel-bobm Bob thanks for weighing in!  Here are my comments just looking at it from the outside, fwiw:

- I have not yet found a test case that actually triggers any 3 second action.   
- Yes I am testing with Loconet decoders, however I can create the issue I observed with the Turnout Tool with no decoders connected whatsoever.  The Turnout Tool shows the state as Inconsistent after clicking the Thrown button one time, and then shows the state as Thrown after clicking the Thrown button again even though the only Loconet traffic is the Switch Request sent by JMRI.
- I use Exact Feedback all the time with the Output Off message disabled.  Generally I am only concerned with the appearance of the turnout on a panel and the behavior of signal logic.  Everything works exactly as I expect for those purposes.  The users of my products are similarly very happy with JMRI handling of Exact Feedback and the related icon animation on panels!

devel-bobm, 2024-03-26:
> I have not yet found a test case that actually triggers any 3 second action.

Not surprising, really.  Goes back to the code originator, whoever that was...

> - Yes I am testing with Loconet decoders, however I can create the issue I observed with the Turnout Tool with no decoders connected whatsoever.

That is a big problem.  Why this happens is problematic.  And worse is that there is no test case that fails for your case!  I do not intend to try to figure out when the code was changed that caused this phenomena, though.

But if you have the problem with physical hardware, that needs to be documented too.  And probably a new issue opened.

> - I use Exact Feedback all the time with the Output Off message disabled.  

But which feedback method do you use - two Sensors, or <OPC_SW_REP>?  It can make a big difference.

> Generally I am only concerned with the appearance of the turnout on a panel and the behavior of signal logic.  Everything works exactly as I expect for those purposes.

Then I guess I am wondering what the problem is.  Sure, the code does not work in the case you show.  But the testcases work...  Shouldn't we be happy?

In case 2, the hardware never reported the <OPC_SW_REP> message, and, then, after a 3 second wait, the <OPC_SW_REQ> gets _retried_, and JMRI _assumes_ that the retried message ran _and_ succeeded, at some unknown-to-me time that has not beed reported.  The _assumption_ that a "programatic retry" will work is problematic to me.  
 
Consider this case, one of a few that occur to me: if the DCC track power is off, the LocoNet message did not get to the DCC track packet stream.  But the DCC packet stream happens to be the _only_ way to get a switch command to a DS54.  So, what makes JMRI decide that the DS54 "must have gotten" the message after the second LocoNet <<OPC_SW_REQ> message, without ever seeing an <OPC_SW_REP> message (if LocoNet reporting) or via two JMRI Sensors?  I do not think that there is anything that JMRI can do to assure the switch moved, to its complete movement.  This is plainly a case where "INCONSISTANT" applies, not Thrown or Closed!

Pugwash1, 2024-03-26:
Try this on for size. This presumes we are using a ds64 or 54 wired and configured correctly.
One motor and 2 microswitchs.
If you send a throw or closed and neither micro switch changes then you have 2 options make an assumption or try again, so it tries again. Still neither micro switch changes, now it makes an assumption  that  since neither switch has changed it must already be in that position. 
If you reconfigure the dsxx to send sensor messages , configure it as two sensor. What happens if the switch is already set?

Pugwash1, 2024-03-26:
Same problem with one sensor.

Pugwash1, 2024-03-26:
Maybe we should look for a piece of code that does an invert on the current consistant state instead of a explicit set to false.

bobjacobsen, 2024-03-26:
I think the three-second timer is a red herring.  It only seems to be invoked when an Uhlenbrock connection is configured. Sorry for the confusion.

bobjacobsen, 2024-03-26:
Sorry to be just catching up.  I'm confused by the behavior of the Turnout Tool vs Turnout Table described above.  Could this just be an issue with how the Turnout Tool (not Table) is displaying the state?

n3ix, 2024-03-26:
@bobjacobsen yes, just the Turnout Tool!  As I posted above, the Turnout Table feedback column is fine.   The Turnout Tool does the right thing on the first button click with no feedback received - it shows Inconsistent, but as I posted in case (2) of my original comment it does the wrong thing when it changes the state to match the button you click the button a second time.

Pugwash1, 2024-03-26:
@bobjacobsen 
For two sensor the issue exists for nce as well. The turnout table will change, never showing inconsistant. If you flip the sensors the turnout table reacts (but never inconsistant) correctly for valid on-off or off-on.
Its futher up the food chain (or a little copy paste went on). Panel editor switchs are similar if you click the switch twice when there is no feedback fired just use internal sensors, then it shows switched when it isnt.

n3ix, 2024-03-26:
@Pugwash1 The Turnout Table shows the Feedback State in the Feedback Column.  I think only the Turnout State can be Inconsistent, the Feedback State is what it is?  Panel Icons do exactly the right thing in my testing so I believe everything is working fine in general and it is only the Turnout Tool that has an issue?

bobjacobsen, 2024-03-26:
I think the problem may be here, in the turnout tools `throwButtonActionPerformed` method:
https://github.com/JMRI/JMRI/blob/4de071895e028c3aff8bdeb874c1a353673ec83c/java/src/jmri/jmrit/simpleturnoutctrl/SimpleTurnoutCtrlFrame.java#L216

```
            if (turnout.getCommandedState() == Turnout.THROWN) {
                nowStateLabel.setText(InstanceManager
                        .turnoutManagerInstance().getThrownText());
            }
```

n3ix, 2024-03-26:
@bobjacobsen looks like this would explain it, and similar for the Closed command code.  The commandState code invoked on the second click essentially overrides the code that has already set the nowStateLabel based on the knownState?

bobjacobsen, 2024-03-26:
I've cleaned those up in the `bobjacobsen/turnout-control-display`.  Could somebody please try that to make sure it fixes the issue?  Thanks.

n3ix, 2024-03-26:
@bobjacobsen happy to test if I can get a .jar, but I haven't been able to build jmri for awhile.

Pugwash1, 2024-03-26:
@bobjacobsen Appears to get the job done.

bobjacobsen, 2024-03-26:
Thank you!  Now in PR #12984

closed 2024-04-30 by n3ix