JMRI/JMRI #12982: 308

wjstix, 2024-03-26:
Issue Type: __Bug__

Hello, I get a 308 error everytime I try to read a new loco , although the programming track is working and I can hear the decoder buzzing and clicking. Sometimes it comes up with MFG ID 255, and reads all CVs as 255. I've tried a new profile, added a PowerPax booster. When I go to Edit>Defaults, no buttons are active. I click on them and save, and when I reboot, they're blank again.

DecoderPro 5.4+R9005ecb4ac
OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.22+7-LTS mixed mode
Windows 11 10.0 amd64

[profile.zip](https://github.com/JMRI/JMRI/files/14751661/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/14751665/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/14751666/systemInfo.txt)

wjstix, 2024-03-27:
> This is almost certainly a DCC hardware problem and NOT a JMRI problem.
>
> Can the DCC system read the decoder when using a throttle, when JMRI is not involved? If it cannot, then the problem is between the command station and the decoder. In that case, anything that JMRI can do will not help, and you should seek help with the DCC system's support people.
>
> If, on the other hand, the throttle can read CVs successfully, but JMRI cannot, then we will need more info.

Hello, thanks for your reply!
I'm only using JMRI for Decoder Pro, connecting a laptop to a dedicated programming track using a Digitrax PR4. I'm not using JMRI software to run the layout or signals etc.
The engines/decoders I've tried to read can all be read and programmed on a separate programming track I have connected to an old Digitrax Zephyr, and I can adjust CVs using OPS programming with the CVP system I use to run the layout. 
Also, I can use the PR4/programming track set-up to read and adjust Digitrax sound decoders using their Soundloader software. It's seems to just be Decoder Pro that has the problem in the last couple of months.
Bill

Breezlys, 2024-03-27:
As suggested (twice!) in your jmriusers thread, and confirmed here by your systeminfo.txt, you have two LocoNet connections defined for your single PR4. I'm not sure if that's causing your issues, but I'd certainly fix the obvious stuff first if for no other reason than to rule it out as a possible contributing factor:

|Connection LocoNet|Digitrax connected via LocoNet PR4 on COM3 (enabled)|
|Connection LocoNet2|Digitrax connected via LocoNet PR4 on COM3 (disabled)|

wjstix, 2024-03-29:
I'm guessing that this problem (everything is apparently set up correctly, but the Defaults page won't allow me to save the button choices, essentially turning everything off) is a new one that hasn't come up before? I've had replies here and in other places, but no one has referenced the actual issue.

Pugwash1, 2024-03-30:
@wjstix there are really no defaults for a pr4 standalone programmer. It's only meant to support service mode programming, The top right should say loconet, if you have 2 connections you must select the correct one, and mode direct byte. Programming on track, bottom right. There have been some  kludges done that allow running on the track for testing, but mostly don't work well and can leave the decoder in an odd state.  No boosters are required.

wjstix, 2024-03-31:
Hi, when I try to check/update settings, I get an error message "Defaults are invalid, do you want to correct them". That's why I figured that must be where the problem is. Bill

closed 2024-08-12 by github-actions[bot]