JMRI/JMRI #12988: DccConsist getConsist Returns Internal ArrayList Instead of Copy

DeanCording, 2024-03-28:
jmri.implementation.DccConsist contains the following code:

```
    protected ArrayList<DccLocoAddress> consistList = null; // A List of Addresses in the consist

   ...

   // get a list of the locomotives in the consist
    @Override
    public ArrayList<DccLocoAddress> getConsistList() {
        return consistList;
    }
```

This passes the actual internal ArrayList to the caller which would allow the caller (such as a script) to modify the consist without it being synchronized with the layout and could cause invalid direction and position information for locos in the consist.

Method should  return `new ArrayList<>(consistList)`

bobjacobsen, 2024-03-28:
Would returning an unmodifiable list break anything?
```
   return java.util.Collections.unmodifiableList(consistList);
```

That way the list is still "live" and will receive updates if any.

DeanCording, 2024-03-28:
Two places where an unmodifiable list might break code is in the Consist Tool and WifiThrottle:

```
jmrit/consisttool/ConsistToolFrame.java:        

ArrayList<LocoAddress> existingConsists = consistManager.getConsistList();

java.util.Collections.sort(existingConsists, new jmri.util.LocoAddressComparator()); // sort the consist list.
```

```
jmrit/withrottle/ConsistController.java:                

ArrayList<DccLocoAddress> conList = consist.getConsistList();

conList.add(conList.remove(index));
```

`new ArrayList<>(consistList)` would be consistent with other classes such as DefaultTransit.

bobjacobsen, 2024-03-28:
Thanks for finding those.  I agree a copy would be best.

DeanCording, 2024-03-28:
The bit of code in WiThrottle/ConsistController is:

```
   private void reorderConsist(String message) {
        Consist consist;
        List<String> headerAndLocos = Arrays.asList(message.split("<:>"));

        if (headerAndLocos.size() < 2) {
            log.warn("reorderConsist missing data in message: {}", message);
            return;
        }

        try {
            List<String> headerData = Arrays.asList(headerAndLocos.get(0).split("<;>"));
            //
            consist = manager.getConsist(stringToDcc(headerData.get(1)));

            List<String> locoData = Arrays.asList(headerAndLocos.get(1).split("<;>"));
            /*
             * Reorder the consistList:
             * For each loco sent, remove it from the consistList
             * and reinsert it at the front of the list.
             */
            for (String loco : locoData) {
                ArrayList<DccLocoAddress> conList = consist.getConsistList();
                int index = conList.indexOf(stringToDcc(loco));
                if (index != -1) {
                    conList.add(conList.remove(index));
                }

            }

        } catch (NullPointerException e) {
            log.warn("reorderConsist error for message: {}", message);
            return;
        }

        writeFile();

    }
```

This is directly manipulating the consistList rather than using appropriate the class methods.

bobjacobsen, 2024-03-29:
Perhaps the best way to deal with these direct accesses is to move them into the ConsistManager as new `sortConsistList` and `moveToTop` methods.

closed 2024-08-12 by github-actions[bot]