JMRI/JMRI #12992: Quiting JMRI with an AutoTrain running fails to stop train before exiting.

DeanCording, 2024-03-30:
When you quit JMRI with a manual throttle in operation, the throttle is set to Stop before JMRI exits so as to prevent runaways.

For trains running as AutoTrain under Dispatcher, the automatic control is not terminated and the throttle is not set to Stop before JMRI exits, potentially leading to runaways.

danielb987, 2024-04-03:
@Pugwash1 
Is this something for you to look at?

Pugwash1, 2024-04-04:
@danielb987 
I don't think so. Dispatcher just uses the standard JMRI throttle. 
If JMRI crashes out do the WiFi throttles all set the speed to zero? Do web throttles set speed to zero? Warrant Throttles?

danielb987, 2024-04-04:
I don't know how these things works. Maybe it's only the throttle GUI that stops the train when JMRI shuts down.

Note that if I understands @DeanCording correctly, the issue is when JMRI exits cleanly, not when it crashes.

danielb987, 2024-04-04:
> I know I spent quite a bit of effort to make sure MQTT throttles closed down safely with manual throttles, so this was a bit of a surprise. 

How does MQTT throttles closes down safely? If I use a manual throttle to run the train, is it the throttle GUI or the MQTT throttle that stops the train when JMRI quits?

DeanCording, 2024-04-04:
On Thursday, 4 April 2024 12:15:54 PM AEST Daniel wrote:
> How does MQTT throttles closes down safely? If I use a manual throttle to
> run the train, is it the throttle GUI or the MQTT throttle that stops the
> train when JMRI quits?

It relies on the controlling throttle to properly dispose of the MQTT throttle 
(which wasn't happening correctly by manual throttle at the time).  The manual 
throttle disposes of the MQTT throttle, and on disposal MQTT throttle cancels 
all of the current throttle commands.

DeanCording, 2024-04-04:
Here is the original issue involving manual throttles #12287

Pugwash1, 2024-04-04:
To do this correctly it has to be in throttlemanager so its behavior is consistant across all platforms.

closed 2024-07-29 by github-actions[bot]

Pugwash1, 2024-07-30:
This was addressed in #13082.