JMRI/JMRI #13001: Java error

ManicMayhem, 2024-04-01:
Hi I am getting a Java error whenever I try and go to preferences. This is on a windows 11 pro version.
JMRI loads OK but crashes out with the following when I try and got to Edit>Preferences.
Any assistance would be much appreciated.
Neil

Unrecoverable error encountered.
This application will quit.

java.util.ServiceConfigurationError: jmri.swing.PreferencesPanel: Provider apps.FileLocationPane could not be instantiated
java.lang.Exception
java.util.ServiceConfigurationError: jmri.swing.PreferencesPanel: Provider apps.FileLocationPane could not be instantiated
java.lang.Exception: java.util.ServiceConfigurationError: jmri.swing.PreferencesPanel: Provider apps.FileLocationPane could not be instantiated
 at jmri.util.exceptionhandler.UncaughtExceptionHandler.uncaughtException(UncaughtExceptionHandler.java:35)
 at java.base/java.lang.ThreadGroup.uncaughtException(Unknown Source)
 at java.base/java.lang.ThreadGroup.uncaughtException(Unknown Source)
 at java.desktop/java.awt.EventDispatchThread.processException(Unknown Source)
 at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
 at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
 at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
 at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
 at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
 at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

bobjacobsen, 2024-04-02:
@ManicMayhem I've not seen an error like this before.  It looks like it's happening inside Java itself, when Java isn't able to load parts of JMRI.  

What version of JMRI are you using?

ManicMayhem, 2024-04-02:
@bobjacobsen 
JMRI version 5.6 and Java version 11.0.22 LTS. 
openjdk version "11.0.22" 2024-01-16 LTS
OpenJDK Runtime Environment (build 11.0.22+12-LTS)
OpenJDK 64-Bit Server VM (build 11.0.22+12-LTS, mixed mode)

I have removed both Java and JMRI a few times and done fresh installs but still keep getting the same result. Tried Java from a couple of different sources as well. :-)

ManicMayhem, 2024-04-02:
@bobjacobsen 
Hi Bob thanks for replying so quickly.
I had installed for all users and tried the second suggestion but to no avail.
As I only use this device for JMRI, my Z21 maintenance DCC-EX and Arduino  stuff I decided to do a complete restore to factory and rebuilt Windows.
Sledgehammer to crack a nut. But reinstalled everything and JMRI is running as sweet as a nut. I ave JMRI on a Pi 5 and my iMac and never had any issues. Jut one of those things. Again many thanks.

klk32003, 2024-04-05:
@ManicMayhem 
Since this is JMRI version 5.6, I would like to ask:
Did you set the "JavaSoft (Oracle) registry keys" option during Java installation as recommended [here](https://www.jmri.org/java/)?

Trucoder786, 2024-04-09:
I think you are not handling Exceptions in your code this may be giving you error

closed 2024-07-08 by github-actions[bot]