JMRI/JMRI #13002: Signal mast create in table then add to panel. The panel xml file signalmasticon litmode is always "false".

Sturbee, 2024-04-02:
Using JMRI version 5.7.5
Signal mast creation is WPRR-1971 and OpenLCB.
Panel xml file has to be edited to change litmode="false" to litmode="true"

bobjacobsen, 2024-04-02:
Could you please upload your XML file?  

You can do that on the [GitHub web interface for this Issue](https://github.com/JMRI/JMRI/issues/13002), but you'll have to either add a .txt to the name or upload a .zip of the file.

Thank you, and sorry for the trouble.

Sturbee, 2024-04-02:
Hi, I will upload the panel xml file shortly.
  
Creating the table signal mast has a UnLit checkbox but when saved to table that litmode flag is not included in the signalmast data.
  
Next add the table signal mast to a panel and the signal mast icon data has litmode=“false” which I edit to “false”

dsand47, 2024-04-02:
litmode is controlled by icon menu.

dsand47, 2024-04-02:
litmode is controlled by icon menu.

[image]

Sturbee, 2024-04-02:
Thanks for pointing out the icon controls the litmode, not the creation of the signal mast.

Suggest removing the UnLit checkbox from the table signal mast create if icon controls the litmode to avoid confusion.

Screenshot.2024-04-02.at.8.52.19.AM.png (view on web)<https://github.com/JMRI/JMRI/assets/17280992/efebd886-62fb-469f-824e-40c1a882ac7a>

closed 2024-06-24 by github-actions[bot]