JMRI/JMRI #13015: Signal Mast / Advanced Accessory command support for DCC-EX

habazut, 2024-04-06:
Today JMRI handles advanced accessory DCC command support for most devices by generating a DCC packet and then tunneling that DCC packet through direct DCC packet support into the DCC packet stream. That does however bypass the logic in the CS.

DCC-EX (DCCpp) has implemented the `<A ...>` accessory command that can handle signal mast or signal head aspects (the terminology is not clear here). It will be in the upcoming release of DCC-EX (5.4.0). It can be tested in the devel releases >=5.2.41. So instead of creating a DCC packet and sending the `<M ...>` command, JMRI should add a signal aspect function to DCCpp and then use that function to send aspect commands instead. JMRI can fallback to sending `<M ...>` if `<A ...>` returns `<X>`.

The functions to overload are probably setAspect or so. The generic implementation is in implementation/DccSignalHead.java and implementation/DccSignalMast.java. Bidib and mqtt seem to implement such a non-generic method.

Regards,
Harald.

danielb987, 2024-04-06:
@mstevetodd 
Is this something for you to look at?

mstevetodd, 2024-04-27:
> @mstevetodd Is this something for you to look at?

I started looking at it but was told that the DCC-EX side is in flux regarding how to handle this, so I'll hold off for now.

mstevetodd, 2024-04-27:
Just to document where I was, my plan:
Add a new "Mast Driver" called "DCC-EX Signal Mast". This would be cloned from "DCC Signal Mast Decoder". 
Replace the "DCCMastAddress" with "Mast ID"
Eliminate "Offset Address" and "Send Count".
Expect the command station to broadcast changes to SignalMasts. So JMRI would send the _request_ to set MastID to AspectID, but not directly change the local JMRI mast aspect. Changes to the JMRI mast would come from handling the broadcast from EX-CS. This would allow EX-CS to veto changes, and also to provide updates for changes made by other clients, automation, etc.
[image]

closed 2024-07-29 by github-actions[bot]