JMRI/JMRI #13017: DispatcherFrame returns internal ArrayList object instead of a copy

DeanCording, 2024-04-07:
Similar to #12988, DispatcherFrame returns internal ArrayList objects instead of making copies.  The code is:

```
     private final List<ActiveTrain> activeTrainsList = new ArrayList<>();  // list of ActiveTrain objects
    protected final List<AllocatedSection> allocatedSections = new ArrayList<>();  // List of AllocatedSection objects

    public List<ActiveTrain> getActiveTrainsList() {
        return activeTrainsList;
    }

    protected List<AllocatedSection> getAllocatedSectionsList() {
        return allocatedSections;
    }
```

Apart from allowing these internal data structures to me manipulated by external code, including scripts, it is also a source of Concurrent Modification Exceptions if these structures are accessed from code in different threads.

For example, this Jython script code is susceptible  to failing with due to a concurrent modification:

```
        DF = jmri.InstanceManager.getDefault(jmri.jmrit.dispatcher.DispatcherFrame)
        trainsList = DF.getActiveTrainsList() #loop thru all trains
        for train in trainsList:
            # do something with the train
```

As the Jython script is run in a different thread from DispatcherFrame, this will fail if DispatcherFrame modifies the list whilst the Jython script is iterating over it (which is how I came across the issue).

I've noticed that a lot of the scripts written by @bobjacobsen uses code like:
 
```
        DF = jmri.InstanceManager.getDefault(jmri.jmrit.dispatcher.DispatcherFrame)
        trainsList = DF.getActiveTrainsList() #loop thru all trains
        for i in range(trainsList.size()):
            train = trainsList.get(i) 
            # do something with the train
```

Whilst this avoids the Concurrent Modification Exception, it has a more subtle bug because the number of objects in the list or the order of those items may change whilst the list is being traversed - those changes may not be detected leading to unstable scripts.

Pugwash1, 2024-04-08:
It would appear one script does update the activeTrainsList, I dont know how welcoming they  be to a change.

The getAllocatedSectionsList is protected so we can do what ever with it.

DeanCording, 2024-04-08:
`It would appear one script does update the activeTrainsList, I dont know how welcoming they be to a change.`

activeTrainsList is a private variable meaning no-one outside of DispatcherFrame should be updating it  (or even accessing it).  If a script in a different thread is updating a private variable then there is a strong possibility that it will crash DispatcherFrame due to triggering concurrent modification exception.

I assume you are talking about DispatcherSystem, it has issues:

    DispatcherSystem/Simulation.py:            for activeTrain in DF.getActiveTrainsList():

This can crash with a concurrent modification exception.

 ```DispatcherSystem/RunDispatch.py:        activeTrainsList = DF.getActiveTrainsList()
DispatcherSystem/RunDispatch.py:        for activeTrain in activeTrainsList:
```
Same as above

```DispatcherSystem/ResetButtonMaster.py:        for activeTrain in DF.getActiveTrainsList():
```
And again

DispatcherSystem/StopDispatcherSystem.py:        for activeTrain in DF.getActiveTrainsList():

And again

```DispatcherSystem/RunDispatch.py:        activeTrainsList = DF.getActiveTrainsList()
DispatcherSystem/RunDispatch.py:        for i in range(0, activeTrainsList.size()) :
DispatcherSystem/RunDispatch.py:            activeTrain = activeTrainsList.get(i)```

This doesn't detect changes to activeTrainsList which can cause subtle hard to detect bugs.

However, DispatcherSystem doesn't update DispatcherFrame.activeTrainsList - it has some similarly named internal variables (activeTrainList & active_trains_list) which it updates.

`The getAllocatedSectionsList is protected so we can do what ever with it.`

No, no, no, that's not how protected variables work.  Protected means that only child classes of DispatcherFrame or classes in the jmri.jmrit.dispatcher package **are trusted** to access allocatedSectionsList, not that there is any sort of protection mechanism.  It certainly doesn't provide any cross thread concurrent access protection.

DispatcherFrame could use java.util.concurrent.CopyOnWriteArrayList to make these variables thread safe.

However, handing out private and protected variables to anyone who asks defeats the whole purpose of private and protected modifiers.

DispatcherFrame should return a clones of the variables instead - ie activeTrainsList.clone()

Pugwash1, 2024-04-09:
I never said you shouldnt make the change, i never said it wasnt the right thing to do, i never asked for a repeat of your arguments.
I re iterate.
 It would appear one script does update the activeTrainsList, I dont know how welcoming they be to a change. 
So you should maybe find out, coordinate updates, be nice to your neighbours.
The  getAllocatedSectionsList is protected so you can do what ever with it.
That means you should be able to update the code without reference to anyone else.

DeanCording, 2024-04-09:
Sorry if I mis-interpreted your comments.

Looking through all of the 'official' scripts, I can't find any that directly update activeTrainsList.  DispatcherSystem looks like it does, but it actually has it's own internal variables called activeTrainsList, or some variant there of.

I can't vouch for any un-official scripts however, so it could be considered a breaking change.

Agreed that allocatedSections should be able to be changed with minimal consequences but still should be considered a breaking change.

I'm happy to make the PR but I thought it might need some more discussion and an agreement on the resolution by those with historical knowledge of the code.

It might also be worthwhile to do an audit of the code for other instances of this.

Pugwash1, 2024-04-09:
No problem
I was running dispatcher (not jython), manual and auto, today using Collections.unmodifiableList() and didn't have any problems.
If you dig you will find, so i personally don't dig if I don't have to....

closed 2024-07-01 by github-actions[bot]