JMRI/JMRI #13037: Train-o-matic lokommander II: f3/f4 handled as int instead of bits

reinout, 2024-04-16:
My decoder got detected as a lokommander II, correctly. I like the customized pages for the various settings!

There is an error on the page for handling the `114` and `115` (and probably `116` too) CV values on the `tOm Function Map tab. 
The default settings are, like many decoders, `F3` for shunting mode (cv 114) and `F4` for switching off accel/decel (cv 115). The interface shows those values as F4 and F8 respectively:

[image]

F4 should be F3 (`4` is binary `001`, so the third function key), likewise F8 should be F4 (`8` as binary `0001`).

When I changed it to F3 and F4 in the interface, the behavior was funny :-) F3 was translated to `3`, so binary `11`, which means that shunting mode was actually enabled with F1 and F2.

reinout, 2024-04-17:
Oh, I forgot to include that I'm using the `5.7.5` release.

danielb987, 2024-04-21:
@reinout 
Welcome to the JMRI project!

@lbijlsma 
Is this something for you to look at?

lbijlsma, 2024-04-21:
Yes, assign it to me and I'll have a look at this issue.

danielb987, 2024-04-21:
Thanks! Done.

lbijlsma, 2024-04-22:
In comparing the decoder manuals for the original Lokommander II firmware and the latest v6 firmware, it looks like the definition for CV114 and CV115 are changed. In the original firmware it is handled in binary and in the v6 as decimal. 
Right now it has a common definition between the firmware versions. I need to separate this out.

@reinout - can you share the firmware version of the decoder and the model of the decode as well? The "Train-O-Matic (tOm)" tab has this information:
- Firmware version
- Product number

lbijlsma, 2024-04-23:
Submitted pull request (#13049) with the changes.

reinout, 2024-04-23:
Firmware version: `6.11.318`.
Product number: `2010212`.

lbijlsma, 2024-04-23:
Interesting because you have the v6 firmware and that should handle it as decimal and not binary (as it was in the v3 firmware). 
That might mean that my fix that I just submitted might not fix it. 
I do have the same decoder (2010212, aka the W8P), so let me look more into this.

reinout, 2024-04-23:
Yes, when looking at the PR, I wondered whether binary and decimal got swapped.
I've got notifications enabled on this issue, so I'll hear it if I need to do more verifying/checking :-)

reinout, 2024-04-23:
Oh, you just changed the comment. Detection of the version: I'll check that in a few minutes.

lbijlsma, 2024-04-23:
I have the same decoder with the exact same firmware and product ID and it gets correctly identified.

CV114, 115 and 116 are on the Brake page, because tOm as all the info in their brake section the decoder manual.

[image]

lbijlsma, 2024-04-23:
[image]

Here is my firmware and version

reinout, 2024-04-23:
My "roster entry" says this:

[image]

tOm function map page says this:

[image]

My initial screenshot was from the brake/stop page:

[image]

I'll try writing the info to the loco again.

lbijlsma, 2024-04-23:
[image]

And this is where CV 114, 115 and 116 are represented. I checked CV114 through a direct CV write to 28 and then read the page and I correctly showed F28.

reinout, 2024-04-23:
Weird... I just did a "read full sheet" on the "brake/stop" page and it correctly detected F3 and F4.
A couple of days ago it switched to F4/F8 upon reading. I'm pretty sure about that.

So... I might have initially had the wrong decoder type selected? I don't remember re-setting the value, but I have been busy with three locomotives the last few days, so I don't fully trust my memory. All were lokommander types, though.

lbijlsma, 2024-04-23:
Could it be that you have different firmware versions in the decoders?

(Toch leuk dat twee Nederlanders hier bezig zijn met een Roemeens decoder in JMRI :-) )

lbijlsma, 2024-04-23:
Because the current implementation for the older firmware version is incorrect and my PR that I submitted earlier today addresses that.

reinout, 2024-04-23:
All four decoders were bought two weeks ago. I do now notice that one of them shows up as a V3 instead of V6, which is wrong. So it is a good bet that the loco I was having the problem on also had that configuration wrong.

Just guessing: I had problems with the latest stable release (m1 macbook) and upgraded to the `5.7.x` not-yet-stable version as it apparently fixed a connection issue on newer macbooks. Well, it *did* fix the connection issue and everything is working like a charm.

But.... The older version might have mis-detected the decoder, which persisted in my configuration. That's my guess.

(En ja, ik vond al dat "Bijlsma" erg Nederlands klonk :-) )

lbijlsma, 2024-04-23:
You might want to re-read the firmware version on that particular decoder and it could be old stock and therefore older firmware. 
I have the tOm programmer so I can do firmware upgrades and I flashed one back to v3 to see that I had it wrong for the v3 decoders, so I'm happy that this was detected and my PR will address that issue.

lbijlsma, 2024-04-23:
Motor control is the same between the two versions. The function functionality is almost a complete rewrite. v6 has a lot more function mapping capabilities. 

Now you have a good way to test my PR, as soon as that is available in a test release, you can test it out. 

And too bad I live to far away, otherwise I would have offered to do a the firmware upgrade.
(I'm in the USA, but modeling Norwegian stuff, you might have seen me on Beneluxspoor with Norwegian posts).

lbijlsma, 2026-10-27:
@danielb987  / @bobjacobsen  this issue as resolved two years ago, this issue can be closed.

closed 2026-10-27 by bobjacobsen