JMRI/JMRI #13044: Communication between MX10 and JMRI (error 306)

SwFw2318, 2024-04-19:
Issue Type: __Bug__

You need a Zimo MX10.

Repeat the CV reading over and over again.

DecoderPro 5.6+R89a87446cb
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 11.0.16.1+1-LTS-1 mixed mode
Windows 11 10.0 amd64

[profile.zip](https://github.com/JMRI/JMRI/files/15043434/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/15043435/session.log)

bobjacobsen, 2024-04-20:
Does the CV read sometimes work? Does it work once and then fail?  Or does it never work?

SwFw2318, 2024-04-21:
It works very badly.
Sometimes writing works (acknowledgement "OK") or sometimes not (program tool error: communication with central processing unit failed (306)).
Some decoders can be read and others cannot.

MX10 Firmware Version V01.30.0560

 have configured the Zimo MX10 in this way:
- System manufacturer: Roco
- System connection: Roco Z21
- IP address/host name: 192.168.1.145 (read from the MX10)
- Connection Prefix: Z
- Connection Name: Z21
- Communication interface: 21105 (cannot be changed)
- Output interval (ms): 250

Default settings for decoder programming:
- Program to: XpressNet
- Mode: Direct Bit

I excuse me for my English.

SwFw2318, 2024-04-21:
The LAN integration of the MX10 in JMRI is missing.
The communication between the Zimo MX10 and JMRI is very poor.

[ZIMO_CAN_Protokoll_4.20_Public.pdf](https://github.com/JMRI/JMRI/files/15051052/ZIMO_CAN_Protokoll_4.20_Public.pdf)

SwFw2318, 2024-04-22:
I synchronize several locomotives (speed) with each other on the main track (PoM), the MX10 is better suited for this.

higaski, 2024-04-25:
I can conform that this ain't working at all. I can connect to an MX10 (01.30.0552) using the Z21 protocol but that's about it. It answers a couple of commands (e.g. reading serial number) but JMRI does not let me open the program window....

closed 2024-07-15 by github-actions[bot]