JMRI/JMRI #13050: Uncought null pointer exception.

habazut, 2024-04-23:
1. Use following config:

```
<?xml version="1.0" encoding="UTF-8"?>
<auxiliary-configuration xmlns="http://www.netbeans.org/ns/auxiliary-configuration/1">
    <connections xmlns="http://jmri.org/xml/schema/auxiliary-configuration/connections-2-9-6.xsd">
        <connection xmlns="" class="jmri.jmrix.dccpp.serial.configurexml.ConnectionConfigXml" disabled="no" manufacturer="DCC++" port="ttyUSB0" reconnectMaxAttempts="100" reconnectMaxInterval="120" speed="115200" systemPrefix="D" turnoutInterval="250" userName="DCC++">
            <options>
                <option>
                    <name>StartUpDelay</name>
                    <value>1.5 seconds</value>
                </option>
            </options>
        </connection>
        <connection xmlns="" address="(none)" class="jmri.jmrix.dccpp.dccppovertcp.configurexml.ConnectionConfigXml" disabled="yes" manufacturer="DCC++" mdnsConfigure="false" port="1235" reconnectMaxAttempts="100" reconnectMaxInterval="120" systemPrefix="D2" turnoutInterval="250" userName="DCC++2">
            <options/>
        </connection>
    </connections>
    <startup xmlns="http://jmri.org/xml/schema/auxiliary-configuration/startup-4-3-5.xsd"/>
</auxiliary-configuration>
```
(second connection is disabled)

2. Start DCC++ over TCP server

3. Connect with DCC++ client to DCC++ TCP server (or simulate with telnet):

```
$ telnet localhost 1235
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
VERSION DCC++ Server 5.7.6plus+haba+20240423T1455Z+Ra49f9a4917
<s>
```

Produces NULL pointer crash like this:

```
     [java] 18:59:44,697 jmri.jmrix.dccpp.dccppovertcp.Server  INFO  - Server: Connection from: /127.0.0.1:58484 [DCCppOverTcpServer]
     [java] 18:59:46,436 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [ClientRxHandler:/127.0.0.1:58484]
     [java] java.lang.NullPointerException: null
     [java] 	at jmri.jmrix.AbstractMRTrafficController.sendMessage(AbstractMRTrafficController.java:346) ~[classes/:?]
     [java] 	at jmri.jmrix.dccpp.DCCppPacketizer.sendDCCppMessage(DCCppPacketizer.java:56) ~[classes/:?]
     [java] 	at jmri.jmrix.dccpp.dccppovertcp.ClientRxHandler.run(ClientRxHandler.java:105) ~[classes/:?]

```
This is due to `xmtRunnable` being `null`, probably it tries to send to the connection that is disabled.

Do not assume that `xmtRunnable` is non-null unless you have a catch for it.

Harald.

danielb987, 2024-04-24:
@bobjacobsen 
Do you have time to take a look at this?

bobjacobsen, 2024-04-24:
I'm not set up to recreate it with a real DCC++ setup, unfortunately,

From the traceback, the NPE is one of those "Should never occur" issues:  We can protect against the NPE with a null check on `xmtRunnable`, but that won't do anything to fix the real error. `xmtRunnable` is set in the `connectPort` method (unless some errors have been logged).  That has to be called before anything can be sent or received on the connection.  So if some DCC++ code is trying to send something before that's complete, that's the true cause of the error. 

The message that's failing is being sent by the `txThread` in `jmri.jmrix.dccpp.dccppovertcp.ClientRxHandler`. In current HEAD that seems to be line 113, not the line 105 cited in the traceback above.  I don't understand DCC++ enough to know why it's sending this message at this point, nor why this could be invoked before the connection is entirely set up.

danielb987, 2024-04-24:
@habazut 
Could you add the complete system console log when this happens?

closed 2024-07-15 by github-actions[bot]