JMRI/JMRI #13062: [Linux] JMRI (java?) freezes when changing loconet preferences from "LocoNet over TCP" to "LocoNet Simulator"

newHeiko, 2024-04-28:
Hi everyone,

I haven't fully debugged this, but in case you can reproduce it: When I connect to a LoconetOverTCP server with JMRI and then attempt to change my connection settings from "LocoNet over TCP" to anything else, including "LocoNet Simulator", JMRI freezes my X Window system. I need to login on a text console and killall -9 java to recover.

Linux 5.15.145
openjdk version "11.0.22" 2024-01-16 LTS
OpenJDK Runtime Environment Zulu11.70+15-CA (build 11.0.22+7-LTS)
OpenJDK 64-Bit Server VM Zulu11.70+15-CA (build 11.0.22+7-LTS, mixed mode)

JMRI as of today's master (4409a9e430b587bb27ebd9e581f80fcfbee6bb38) back all the way to at least v5.2 (further back failed to load my preferences)

Steps to reproduce:

1. Remove ~/.jmri
2. Start DecoderPro
3. In the Wizard, select System Manufacturer "Digitrax", System Connection "LocoNet over TCP", enter correct server address
4. Finish Wizard
5. verify JMRI connects to your LoconetOverTCP server
6. Select Edit -> Preferences
7. Attempt to change System Connection from "LocoNet over TCP" to anything else

Note that if the LoconetOverTCP server is down/unreachable, DecoderPro will load the "Edit Connections" just fine and allow me to change the settings.

The logfile seems to point to some exception mishandling in stopping the LoconetOverTCP connection, see attached for one example. As I said, I have barely started into debugging this, so this is a random poke in the dark.
[session.log](https://github.com/JMRI/JMRI/files/15143515/session.log)

Heiko

P.S: I don't think I'll be having a lot of time between now and end of July (I didn't even plan to do anything on JMRI today), but if either one of you sees the issue immediately or I come back to this in August, I'd like to have it documented :)

bobjacobsen, 2024-04-28:
It's really helpful to get a thread dump when JMRI hangs like this.  If you're running from the console, you can get it with a ctrl-\ on the appropriate window.  Otherwise, it's a SIGQUIT to the Java process.

newHeiko, 2024-04-28:
Not a problem, here you go:

[stacktrace.log](https://github.com/JMRI/JMRI/files/15143841/stacktrace.log)

I have also added the session.log from the same session:
[session.log](https://github.com/JMRI/JMRI/files/15143842/session.log)

Heiko

bobjacobsen, 2024-04-28:
Thank you!  It's hanging up waiting for the receive thread to terminate in LnPacketizer.  

Could you try something for me?  it's just a temporary solution, but it would be good to confirm this is what's actually happening.

Line 483 in `java/src/jmri/jmrix/loconet/LnPacketizer.java` should be:

```
                rcvThread.join();
```

Could you add an argument of 150 to this `join` call so that it looks like:

```
                rcvThread.join(150);
```

If that works, it'll confirm where the problem is.  

Thank you!

newHeiko, 2024-04-28:
Thank you!

Yes, that works. (I put 1500 in and it took about 1.5s for JMRI to become responsive again.)

```
      [java] 00:00:35,186 x.loconet.LocoNetThrottledTransmitter DEBUG - item requests shutdown [LocoNetThrottledTransmitter]
     [java] 00:00:36,776 jmri.configurexml.ConfigXmlManager    DEBUG - register jmri.jmrix.loconet.hexfile.ConnectionConfig@1e56f9d1 adapter jmri.jmrix.loconet.hexfile.configurexml.ConnectionConfigXml [AWT-EventQueue-0]

```

Heiko

danielb987, 2024-05-05:
Closing this PR since it's solved by #13067.

closed 2024-05-05 by danielb987