JMRI/JMRI #13068: Webserver doesn't render Layout Editor turnout or diagonal line correctly

n3ix, 2024-04-30:
Running latest JMRI 5.7.6 build.  Comparison below shows Layout Editor on left and Web Server (using Panels, not Frames) on right.  All track segments are set to Mainline.  Unoccupied track is White, Occupied track is Red, Alternate Color (for Dispatcher) is Blue.  Block boundaries are shown in Yellow.  The turnout and the track segments connected to all three legs are in the same block.  

The first set of images compare the unoccupied turnout renderings and show that the line width of the turnout legs in the Web Server version does not match the connected track segments.

The second set of images shows an issue with the diagonal track segment connected to the diverging turnout leg.  With a blue line on black background the pixels on the edges of the diagonal segment should vary between blue and black.  However there are light colored pixels on the edges.  There is also a faint outline of the non-connected portion of the diverging leg.  I included a zoom-in image 
 to show the issues better.  I tried 2 different browsers with similar result.  

Image attached along with simple panel file used to create it.
[webserver issue demo.zip](https://github.com/JMRI/JMRI/files/15171796/webserver.issue.demo.zip)

danielb987, 2024-05-03:
@mstevetodd 
Is this something for you to take a look at?

dsand47, 2024-05-03:
Layout editor turnout legs use the "mainline" attribute of the attached track segment.

A Layout Editor turnout is not an icon.  It is a set of lines.  If I recall correctly, a standard turnout has 5 lines.

mstevetodd, 2024-05-03:
> @mstevetodd Is this something for you to take a look at?

Yes, I will take a look.

mstevetodd, 2024-05-03:
@n3ix I added icons to set the occupancy, but I'm not sure how to set the alternate color. Can you advise?

I've fixed the track size issue and it works with the white and red.
I think the issue with the faint outline is a byproduct of the browser's anti-aliasing. To "fix" would require turning off the anti-aliasing, or increasing the width of the erased track, neither of which I'm inclined to do.

dsand47, 2024-05-03:
> @n3ix I added icons to set the occupancy, but I'm not sure how to set the alternate color. Can you advise?
> 

@mstevetodd 

The alternate track color is used by Dispatcher and Entry/Exit to indicate allocated blocks.  The LogixNG Block action includes an option to set the alternate track color.

```
LogixNG: Reserve Block
    ConditionalNG: Reserve
        ! A
            If Then Else. Execute on change
                ? If
                    Sensor reserve-block is Inactive
                ! Then
                    Set block "block-xyz" to Alternate Color Off
                ! Else
                    Set block "block-xyz" to Alternate Color On
```

mstevetodd, 2024-05-03:
@dsand47 

Thanks. I was looking for something in the UI which apparently doesn't exist.

dsand47, 2024-05-03:
The track colors, including the alternate track color, work.  There is still an issue with the width of turnout legs.

This image from the JMRI panel shows the narrow diverging leg of turnout 2 based on the attached dashed track segment.

[image]

The same view from the web page has a wide diverging leg.

[image]

Note:  The extra icons on the web view are caused by a script setting the icons not visible after the panel is loaded.

mstevetodd, 2024-05-03:
@dsand47 I could look at that one as well, but I'd need an example panel which shows this behavior. There are quite a few attributes that go into the track color and sizing.

danielb987, 2024-05-04:
@dsand47 
> Note:  The extra icons on the web view are caused by a script setting the icons not visible after the panel is loaded.

Do you mean that if an icon on the panel is set to hidden, the web panel doesn't follow?

I have a similar issue I'm thinking to solve, but I haven't had time for it yet.

mstevetodd, 2024-05-04:
> Do you mean that if an icon on the panel is set to hidden, the web panel doesn't follow?

The web panels get live updates via the json server, which only sends changes to the "state" of objects, not attributes. Refreshing the page re-retrieves the full xml for the panel, so attribute changes come along with that, but at quite a cost.

dsand47, 2024-05-04:
This issue is not totally resolved.

The attached has a simple example of the diverging leg using the wrong width.  

[image]

There is also an example of changing the hidden state of icons using LogixNG,

[hideshowtest.txt](https://github.com/JMRI/JMRI/files/15211092/hideshowtest.txt)

mstevetodd, 2024-05-04:
@dsand47 I still need a panel example. No way I can guess from an image what settings you might have chosen for each track, block, turnout leg, etc.

dsand47, 2024-05-04:
> @dsand47 I still need a panel example. No way I can guess from an image what settings you might have chosen for each track, block, turnout leg, etc.

The sample was attached to my previous posting.  hideshowtest.txt (change to xml).

mstevetodd, 2024-05-04:
Thanks, @dsand47 
I can duplicate the issue. But the fix is going to require rearranging several of the current checks and some dedicated time to insure I don't break other usages. Not sure when that will happen.

commit 1bf2c9ec (2024-05-04): web panels: more fixes for track widths and colors, closes #13068
commit 402f5fe8 (2024-05-04): Merge pull request #13076 from mstevetodd/master

web panels: more fixes for track widths and colors, closes #13068
PR #13081 (2024-05-06): web panels: rewrite turnout color and width logic
closes #13068