JMRI/JMRI #13073: Signal Mast Auto Generated Logic ignores Supporting Turnout Invert option

n3ix, 2024-05-02:
A turnout is involved in the logic for a signal mast.  A Supporting Turnout is added to the Layout Editor turnout.  The auto generated turnouts for the Signal Mast logic now also include the Supporting Turnout.   However the required state of the Supporting Turnout always matches state of the base turnout independent of whether the "Invert Second Turnout" checkbox is enabled.  When Invert Second Turnout is checked an impossible condition is created in the Signal Logic.

The attched example shows turnout "UTJ E M-P" with supporting turnout "UTJ E M-P Reverse".  The Invert Second Turnout box is checked.  Auto Generated Signal Mast logic requires both UTJ E M-P and UTJ E M-P Reverse to be Thrown, which is impossible.

[image]

dsand47, 2024-05-02:
Robin,

Can you supply an image of the track work in this location.  I am having trouble visualizing why the "also throw" turnout needs to be inverted.

Dave Sand

image.png (view on web) <https://github.com/JMRI/JMRI/assets/14222051/1bbd8b69-5537-432c-9422-3a9e0313d6c1>

n3ix, 2024-05-02:
Dave,

 

In this case that turnout output goes to a device that controls a Relay.  The relay supplies the correct DCC polarity power to a crossing.  It was easier to use the Invert checkbox than to rewire the relay 😊  I worked around this but just unchecking “Use Turnout details from Layout Editor” option in the signal mast logic.  

 

I don’t know why Auto_Gen needs to pull in supporting turnouts, but if it does I think it ought to process the Invert selection?

 

Robin

dsand47, 2024-05-02:
Robin,

Since the second turnout (actually just an output event) is not a turnout in the panel sense, then it should be controlled by Logix or LogixNG.  

Supporting turnouts are frequently used in crossovers to drive two switch machines.  That is probably the basis for the logic.

Dave Sand

n3ix, 2024-05-03:
Dave,

 

Ok, I guess there is my way and the JMRI way 😊  Seemed like this was an easy way to get it done, but I’ll go ahead and add LogixNG to invert the turnout instead.   In my world this still a bug though <g>

 

Robin

dsand47, 2024-05-03:
Robin,

Here is an easy override for the signal mast logic.

[image]

n3ix, 2024-05-05:
Dave,

 

I didn’t know that manually including the opposite state would cancel the auto-gen.  Thanks!

 

Robin

 

Screenshot.2024-05-02.at.10.44.28.PM.png (view on web) <https://github.com/JMRI/JMRI/assets/17280992/ddeba2a7-9aaa-492a-b2e2-cc011ee57d7d>

closed 2024-07-29 by github-actions[bot]