JMRI/JMRI #13079: Java Exception

VegasJeff, 2024-05-04:
Hi,
I downloaded JMRI 5.6 and tried to install over 4.26 and failed. I aborted as I had 4.26 running at the time. I even restarted my computer and tried install again. It completed, but when I try to run decoder pro, I get a Java Virtual machine launcher window and error saying "A java exception has occurred. I have uninstalled JMRI and retried the install, but still get the Java message.

I have no idea where to go from here.

Thanks

danielb987, 2024-05-05:
Have you installed Java 11? JMRI 5.6 doesn't work with Java 8.

See this page:
https://www.jmri.org/java/

VegasJeff, 2024-05-05:
Yes, or at least I'm 98% sure. I will double check after work today.
I was able to run 4.2 and do remember it making me installing the
latest version of Java.

Jeff

danielb987, 2024-05-05:
Note that if you go to https://www.java.com/ you will NOT get Java 11 but Java 8.

VegasJeff, 2024-05-05:
Ok, You were right. I had Java 8. I just pulled down Ver 11.

Thx

closed 2024-06-24 by bobjacobsen