JMRI/JMRI #13093: Z21 locos per connection limit is 16

habazut, 2024-05-11:
As stated in the Z21 LAN Protocol specification (section 4, "Driving" under Figure 2) the Z21 has a FIFO with 16 entries where the Locos are stored that get notified for changes. If I look at the JMRI source code and read it correct, all loco commands are sent over the same connection so it is to be expected that from JMRI controlled loco #17 the #1 loco does not any longer gets notifications when the #1 loco is changed by a Multimaus or from the Z21 App.

The solution would be to open more "connections" from JMRI which in the UDP case would mean to use another UDP source port.

I have not verified this by experiment but someone who actually owns a Z21 may do so.

Regards,
Harald.

danielb987, 2024-05-12:
> As stated in the Z21 LAN Protocol specification (section 4, "Driving" under Figure 2) the Z21 has a FIFO with 16 entries where the Locos are stored that get notified for changes. 

Is this FIFO per connection or one FIFO for all the connections? If the latter, then it doesn't help to open many connections.

habazut, 2024-05-15:
> In order to keep network traffic within reasonable limits, a maximum of 16 locomotive 
> addresses per client can be subscribed to (FIFO).

(from PDF)

As this is UDP it is hopefully per client IP/port tuple as there is no other way to define a client identity in UDP. As UDP is connectionless there is no real connection. I wonder if using one InetSocketAddress per throttle object would do the trick. Any non-thottle data could use any of these or an own socket.

Harald.

closed 2024-08-05 by github-actions[bot]