JMRI/JMRI #13110: Webpanel fails to show bumper

n3ix, 2024-05-21:
When a bumper decoration is defined at an anchor where 2 track segments connect, LayoutEditor places the bumper definition in one of the two track segments that connect at the anchor.    If the track segment that happens to contain the bumper is later changed to hidden, LayoutEditor shows the bumper.  However Webpanel no longer shows the bumper even though the other track segment connected to the anchor is visible.  (LayoutEditor shows the bumper even if both connecting track segments are hidden)

Panel file attached.
[webpanel bumper issue.zip](https://github.com/JMRI/JMRI/files/15392966/webpanel.bumper.issue.zip)

[image]

danielb987, 2024-05-21:
@mstevetodd 
Is this something for you to take a look at?

mstevetodd, 2024-05-21:
I'd prefer someone else take a look, maybe whoever added the other decorations to JMRI and the web panel.

n3ix, 2024-07-08:
Steve looks like nobody has picked this up.  Any suggestions on who I might reach out to?

mstevetodd, 2024-07-08:
> Steve looks like nobody has picked this up. Any suggestions on who I might reach out to?

Sorry, I don't recall who added the other decorations to LE and to the web panels.

n3ix, 2024-07-08:
Ok thanks.   FYI LE renders the bumper whether its track segment happens to be visible or hidden.  It's just the WebPanel that doesn't show the bumper when its track segment is hidden, even though the other segment connected to the anchor is visible.  Since the user has no direct control over whether LE assigns the bumper to the first track segment or the second track segment,, currently the only workaround appears to be directly editing the XML.

closed 2024-10-07 by github-actions[bot]