JMRI/JMRI #13111: System frozen when file open dialog opens behind "always on top" C/MRI Monitor window (In Windows 11) (5.7.7)

BCRodich, 2024-05-21:
I had the C/MRI Monitor open (with always on top selected) and tried to open a panel file.  When the File Open Dialog opened, it was completely behind the C/MRI monitor so I could not access it even though that was where focus was.  I needed Task Manager to kill all java processes to shut down JMRI.  This was on a Windows 11, latest JMRI, (5.7.7) and as far as I know the latest Java version.

icklesteve, 2024-05-23:
Hi @BCRodich , thanks for reporting.

I can reproduce this with the JMRI Console Frame Always on Top selected, see #12280 for a previous similar issue.

To prove this is the same cause, I'm playing with tweaks to the File Chooser Dialog, which when supplied with a parent window and modality set to Document local behave as expected.
Will tidy up into a PR,

Steve.

icklesteve, 2024-05-23:
Hi @KenC57 

The Window could not be moved, minimised or accessed as there is an application-wide Modal Dialog open, which restricts access to all application windows until it is closed.
The Dialog could not be easily closed as the always on top window is in front of it.

The fix is to both change the Dialog from the default Application-wide Modal to Document Modal, 
and provide the dialog with a window to stay modal to.
This enables the always on top window to be accessed / moved / resized etc. 
while the Dialog now only disallows access to the window that launched the Dialog.

Hth, Steve.