JMRI/JMRI #13113: JMRI 5.6 Panel Pro does not remember  window positions after config selection on  Pi5 Pi Os Bookworm but does on PC/Win11

Nik4472, 2024-05-22:
JMRI 5.6 Panel Pro

Pi5 Pi OS Bookworm: If  profile selector enabled in Edit/Preferences/Config Profiles and Start Up has Open windows  options then window positions are not remembered from last click on Panel Pro Quit.

Windows 11 PC: If  profile selector enabled in Edit/Preferences/Config Profiles and Start Up has Open windows  options then window positions are remembered from last click on Panel Pro Quit.

bobjacobsen, 2024-05-22:
For some background, see [this jmri-developers thread](https://jmri-developers.groups.io/g/jmri/topic/105899617#msg9999).  Just background; there's no solution there.

This seems to be related to the use of Wayland.

Nik4472, 2024-05-22:
> positions after config selection on  Pi5 Pi Os Bookworm but does on
> PC/Win11 (Issue #13113)

Hi Bob,

Many thanks.

I should have mentioned I had swapped from X11 to Wayland (because of a 
problem with drop down menus in JMRI 5.6 Panel Pro when using X11 over 
Real VNC).

Regards

Nick

<https://jmri-developers.groups.io/g/jmri/topic/105899617#msg9999> . 
<https://github.com/JMRI/JMRI/issues/13113#issuecomment-2124462193> , or 
unsubscribe

bobjacobsen, 2024-07-08:
See [jmri-developers discussion](https://jmri-developers.groups.io/g/jmri/topic/105899617)

mstevetodd, 2024-08-23:
Wayland compositor "Wayfire" (the default) intentionally does not allow an app to set the initial position of its windows.
However, raspi-config now has an option to select the "labwc" compositor, which does allow the app to set the initial position of windows. I have changed my RPi-JMRI image to use labwc with no issues. 
@Nik4472 I suggest you change yours as well.

closed 2024-12-16 by github-actions[bot]