JMRI/JMRI #13114: Extra "Where Used" menu item?

jerryg2003, 2024-05-23:
Version 5.7.2, Windows 11

This command selection is as expected:

`Tools==>Where Used
`
but also have:

`Tools==>LogixNG==>Where Used
`
...with a slightly different popup window display, missing the help line.

Is this intended?

danielb987, 2024-05-31:
The reason for two different "Where Used" is that the first is generic for JMRI but it only shows the LogixNG action/expression that uses a bean.

The second "Where Used" is specific to LogixNG and have the benefit that it shows more detailed information about where a bean is used. The LogixNG, the ConditonalNG, and so on. It makes it much easier to find the action/expression that actually uses the bean.

The reason the second "Where Used" doesn't have help is that I'm not good at writing documentation. So I focus on making the code work. When, if someone wants, it's easy to add the documentation too.

dsand47, 2024-06-01:
Daniel,

Jerry was referring to the window help references.

PR #13152 updates the missing window help references.

Dave Sand

The reason the second "Where Used" doesn't have help is that I'm not good at writing documentation. So I focus on making the code work. Then, if someone wants, it's easy to add the documentation too.

jerryg2003, 2024-06-01:
So Tools==>LoxigNG==>Where Used provides different capability from Tools==>Where Used.  According to @danielb987, it provides more details about LNGs so maybe the menu item should have a different name, e.g.

Tools==>LogixNG==>Where Used (Detail)

closed 2024-06-02 by danielb987