JMRI/JMRI #13117: Power on, sprog as command station, no track power

renaudtrain, 2024-05-24:
Hi,

I have a problem with both my SPROG III modules:
On JMRI, In operation mode, with power off, both SPROG leds are on, stable.
When I turn the power on, The USB side LED is off and the power side LED is blinking. There is no power in the track (a single piece of track for that matter).

Any idea?

Thanks
Renaud

danielb987, 2024-05-24:
@renaudtrain 
Welcome to the JMRI project!

I suggest that you ask that question on the JMRI users group. There are many more people on that list to help you, than reading these issues.
Browse or [join ](https://groups.io/g/jmriusers/join)the [JMRI Users Groups.io](https://groups.io/g/jmriusers/topics) Group

renaudtrain, 2024-05-24:
Will do. Thanks a lot. I thought I was there.

danielb987, 2024-05-29:
@renaudtrain 
Is this issue completed? If so, it would be good if it could be closed.
Thanks.

PUlvestad, 2024-06-18:
Renaud,
I recommend you join the Sprog-DCC users group.
https://groups.io/g/sprog-dcc
Peter

closed 2024-09-09 by github-actions[bot]