JMRI/JMRI #13128: Creating a crossover in Layout Editor causes 2nd turnout to stack on top of first

Mike-Muha, 2024-05-26:
Issue Type: __Bug__

Creating a crossover in Layout Editor causes 2nd turnout to stack on top of first. I'm running JMRI 5.7.7.

1. In Layout Editor, create a right-handed turnout.
2. Create another right-handed turnout that's flipped 180 degrees.
3. Move the second turnout so two red diverging point rectangles overlap, as if you were creating a crossover.
4. Release the turnout.

Result: The second turnout flips 180 degrees and lays directly on top (or under?) the first turnout . The only way to tell is to move the green anchor point rectangle.

Thanks!
Mike Muha
mike.muha@NordicSkiRacer.com

PanelPro 5.7.7+R5cf2ea9d7f
OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.19+7-LTS mixed mode
Windows 11 10.0 amd64

[profile.zip](https://github.com/JMRI/JMRI/files/15448623/profile.zip)

danielb987, 2024-05-26:
@Mike-Muha 
Welcome to the JMRI project!

@dsand47 
What's your view of this issue?

dsand47, 2024-05-26:
There is a subtle bug when trying to use the drag and drop feature when both turnouts are connecting on diverging legs.  Since crossover are normally added using the crossover option, this will be a very low priority.

If separate turnouts are really required, for now connect the diverging legs using a manual track segment.

Mike-Muha, 2024-05-27:
There's an alternative method: if I add an anchor point and a track segment to one of the other turnout legs of the second turnout, I can successfully connect the diverging legs of the two turnouts.

I'm not sure how the crossovers work when I'm using LCC to drive the turnouts, thus my use of two turnouts instead of the X-over...

dsand47, 2024-05-27:
> There's an alternative method: if I add an anchor point and a track segment to one of the other turnout legs of the second turnout, I can successfully connect the diverging legs of the two [...]

That is more work than just drawing the track segment between the two diverging legs.  You need to separate the turnouts in any case to assign a block to the track segment.  I am assuming you are using signal mast logic or Dispatcher at some point.

I use a single turnout address for crossovers.  The turnout events drive a single SMD-8 line.  The SMD-8 connection is daisy chained to two Tortoise switch machines.

dsand47, 2024-05-28:
The code is working properly.

However, the Layout Editor documentation needs to be enhanced.

The first item is to expand the content about the "turnout circle size".  The turnout cycle size is also the "hot spot rectangle" size.  This controls the area of the panel that responds to mouse actions.  For example, dragging an anchor point, moving the connection points on a turnout, displaying the right click context menu, etc.  Reducing the turnout circle size makes it easier to select or grab an item in a crowded area.

The second item is that there needs to be section on the "turnout drag and drop automatic connection" feature.  This is normally used to build yard ladders.  This feature requires that there is only one pair of overlapping unused (red/magenta) turnout connection points.  When both turnouts are LH and/or RH, the second turnout will be rotated.

The previous paragraph has implications for creating a crossover using a pair of regular turnouts.  The first one is that the second turnout cannot be pre-rotated.  This avoids the overlapping connection point restriction but results in the second turnout rotating back.  The second implication is that the A and B connections of the first turnout need to have track segments already attached so that the connections are now occupied or the A-B length of both turnouts are different taking into consideration the hot spot rectangle size.  Changing the length of the C leg of either turnout can also separate the A and B connections.

closed 2024-06-01 by danielb987