JMRI/JMRI #13135: Increase Throttle F-keys to include  F29, F30 & F31

Kcsmith0708, 2024-05-28:
Issue Type: __Bug__

JMRI v5.7.7
[systemInfo.txt](https://github.com/JMRI/JMRI/files/15475079/systemInfo.txt)
[session.log](https://github.com/JMRI/JMRI/files/15475081/session.log)

In the next DCC-EX Command Station v5.4 release we allow throttles to use F-key F29 - F31 to set the PWM Frequency of a 
DC analogue track addresses from 1 to 10239. i.e. setting a specific Cab address to a specific PWM frequency.

Currently I can manually edit the Roster.xml file and increase the maxFnNum="31" F-keys to 31 and the f-key buttons do display on the 'Funtion Lables' pane and on the PC Throttles.

but JMRI .py python scripts don't allow F-keys F29-F31 settings to compile and run 
although F0-F28 work fine.

# Set the Analogue Cab Function Keys 
 self.throttleA.setF0 (True)   # F0-key On DC Throttle [x] Shunting mode
#self.throttleA.setF29 (True)  # F29-key On 490Hz Frequency 
 self.throttleA.setF30 (True)  # F30-key On 3400Hz Frequency
#self.throttleA.setF31 (True)  # F31-key On 62500Hz Frequency

JMRI System Console  Error

874 jmrix.dccpp.DCCppSerialPortController DEBUG - No Flow Control or Buffer Check [serial.SerialDCCppPacketizer Transmit thread]
874 jmri.jmrit.automat.AbstractAutomaton  WARN  - Unexpected Exception ends AbstractAutomaton thread [__builtin__$BackAndForthTimed$59]
org.python.core.PyException: AttributeError: 'jmri.jmrix.dccpp.DCCppThrottle' object has no attribute 'setF30'
    at org.python.core.PyException.fillInStackTrace(PyException.java:72) ~[jython-standalone-2.7.2.jar:2.7.2]
    at java.lang.Throwable.<init>(Unknown Source) ~[?:?]
    at jmri.jmrit.automat.AbstractAutomaton.run(AbstractAutomaton.java:139) [jmri.jar:?]
    at java.lang.Thread.run(Unknown Source) [?:?] 

Is it possible for a future release of JMRI to allow for throttle.setF to include F29 F30 F31 ?

Regards
Kevin C Smith
DCC-EX Dev

PanelPro 5.7.7+R5cf2ea9d7f
Eclipse OpenJ9 VM Eclipse OpenJ9 openj9-0.36.1 JRE 11 Windows 11 amd64-64-Bit Compressed References 20230214_572 (JIT enabled, AOT enabled)
OpenJ9   - 0592661e4
OMR      - f491bbf6f
JCL      - 4e060ceb0b based on jdk-11.0.18+10
Windows 11 10.0 amd64

Kcsmith0708, 2024-05-28:
JMRI .py scripting New feature request

Allow self.throttle.setF   F29 F30 F31 keys to be used in scripts.

icklesteve, 2024-05-28:
Hi @Kcsmith0708 

Please see https://www.jmri.org/JavaDoc/doc/jmri/Throttle.html#setFunction(int,boolean)

e.g. self.throttle.setFunction(29, True)

The setF1 .. setF28 methods just redirect to the setFunction(int,boolean) method.

Who knows what the maximum number of functions will be set at in 20 years from now, however by using setFunction(int,boolean) it's not a problem to add new ones as required,

Steve.

Kcsmith0708, 2024-05-28:
Hi Steve,

  self.throttle.setFunction(nn, True) works to set our F29 & greater function keys in the .py scripts
 
 Super fast JMRI Support, as usual
 
 Thanks,
 Kevin

danielb987, 2024-05-29:
@Kcsmith0708 
Is this issue completed? If so, it would be good if it could be closed.
Thanks.

Kcsmith0708, 2024-05-30:
#13135 Issue Corrected and Completed
Thx