JMRI/JMRI #13159: Bug: Speed-Cat not responding.

hendriksen-mark, 2024-06-02:
In decoderpro the speedo is not responding to a speed-cat from KPF-Zeller.
If i run jmri in Apache Netbeans the speed-cat does work.

I tried different versions of JMRI also test build but non of them work.
Only if i run in Apache Netbeans.

bobjacobsen, 2024-06-02:
Is there anything in the System Console?  That's found in the Help menu.

hendriksen-mark, 2024-06-02:
I have put it in a pastbin because it is big

https://pastebin.com/NbjVpz20

hendriksen-mark, 2024-06-02:
Some extra info, i have tied on macos14 and macos10.14

bobjacobsen, 2024-06-02:
Thanks.  I don't see anything unusual in that log.  

It's a bit mysterious that it would work under NetBeans but not while running standalone.  The Bachrus code seems to be opening the port just fine, but it's not receiving any input.

Perhaps those are using different preference files?  It might be useful to double check that the connection and default preferences are set the same when running the two ways.

Does your DCC++ connection work well when running standalone?

hendriksen-mark, 2024-06-02:
The config and profile is the same in both.
The DCC++ works good i can control trains and program decoders.

this is the log from netbeans.
https://pastebin.com/iVxeTqyF

hendriksen-mark, 2024-06-02:
In version 5.1.1 just after the introduction of KPF-Zeller Speec-cat it works.

hendriksen-mark, 2024-06-02:
production release 5.4 works.
From release 5.5.1 it does not work.

bobjacobsen, 2024-06-02:
> production release 5.4 works.
> From release 5.5.1 it does not work.

Thanks for checking all that.  That's a big clue.  Between those two, the serial library we use was modified.  Could you put the `lib/purejavacomm-1.0.5.jar` file from 5.4 into your JMRI install to see if that fixes the problem?

The difference between the two is meant to just be the addition of Apple Silicon support.  It shouldn't have caused a problem for your Intel Mac.  But perhaps it did in some unexpected way.  (It's been working for other kinds of connections on lots of other Mac installations, including mine)

hendriksen-mark, 2024-06-02:
with `lib/purejavacomm-1.0.5.jar` from 5.4 it works in 5.7.7

bobjacobsen, 2024-06-02:
@danielb987 This is asking about the jmri.jmrix.bachrus.kpfserialdriver.SerialDriverAdapter class.  We haven't migrated this to the new fazecast communications library yet. It uses the serial listener approach instead of the port stream I/O approach.  I'm not familiar with the listener modifications.  Is this something you could take a look at?

danielb987, 2024-06-03:
@bobjacobsen 
I assume you talk about PR #13058. I have looked at the PR and I don't see anything that could explain this. My PR changes the methods `handlePortNotFound()` and `activatePort()` to static methods and therefore also adds the parameter `systemPrefix` to these.

And it also adds some wrapper methods like `setRTS()` that calls `this.serialPort.setRTS();`

Since @hendriksen-mark solved the issue by restoring the old `purejavacomm-1.0.5.jar`, that lib seems to be the culprit to me.

bobjacobsen, 2024-06-03:
Sorry, I wasn't clear.  

Since this is a problem with the old PJC library that we're migrating away from, I thought perhaps the next step in resolving this is to migrate this KPF system connection to the new fazecast library.

I thought I remembered that you'd migrated some other system that uses the serial listeners instead of the stream I/O.  I haven't done that, so if you had done one, it might be easier for you to do this one.

Again, sorry for the confusion.

danielb987, 2024-06-04:
> Since this is a problem with the old PJC library that we're migrating away from, I thought perhaps the next step in resolving this is to migrate this KPF system connection to the new fazecast library.

I'm looking into it.

danielb987, 2024-06-04:
> Since this is a problem with the old PJC library that we're migrating away from, I thought perhaps the next step in resolving this is to migrate this KPF system connection to the new fazecast library.

See PR #13163.
I have added a PureJavaComm emulator, `jmri.jmrix.purejavacomm`, that uses `jmri.jmrix.AbstractSerialPortController` to replace PureJavaComm with com.fazecast.jSerialComm.

Jenkins build is here: https://builds.jmri.org/jenkins/job/testreleases/job/danielb987/

I have not tested this with any hardware and I'm not sure I understand how it should be, but hopefully we can get this to work.

Another question is copyright. I have looked at the PureJavaComm source code and I have taken some minor parts from that project. But I'm not sure how to do with copyright notes. The exception is the SerialPortEvent class which i copied entirely from PureJavaComm with the copyright statement.

bobjacobsen, 2024-06-05:
@danielb987 Thank you!  That looks good.

@hendriksen-mark Could you please try a JMRI installer that Daniel created?  They're at https://builds.jmri.org/jenkins/job/testreleases/job/danielb987/  Thank you!

hendriksen-mark, 2024-06-05:
Still does not work.
DCC++ can connect and be controlled but the speedo is not able to connect.
In the log is a ERROR.
```
19:41:44,580 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]
19:41:44,602 apps.util.Log4JUtil                   INFO  - This log is stored in file: /Users/markhendriksen/Library/Preferences/JMRI/log/session.log [main]
19:41:44,602 apps.util.Log4JUtil                   INFO  - This log is appended to file: /Users/markhendriksen/Library/Preferences/JMRI/log/messages.log [main]
19:41:44,610 apps.AppsBase                         INFO  - DecoderPro version 5.7.7plus+jenkins+20240604T1613Z+R5e08a405b3 starts under Java 11.0.23 on Mac OS X x86_64 v14.5 at Wed Jun 05 19:41:44 CEST 2024 [main]
19:41:44,735 apps.gui3.Apps3                       INFO  - Starting with profile Mijn_JMRI_Spoorweg.3f010714 [main]
19:41:45,955 jmri.util.node.NodeIdentity           INFO  - Using e0343de8-53a7-40bf-aa0c-35837e8357bb as the JMRI storage identity for profile id 3f010714 [AWT-EventQueue-0]
19:41:46,008 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "DCC++" [main]
19:41:46,056 jmri.jmrix.dccpp.serial.DCCppAdapter  INFO  - Connecting DCC++ to tty.usbmodem142201 Arduino Mega 2560 (Dial-In) [main]
19:41:46,058 jmri.jmrix.dccpp.serial.DCCppAdapter  INFO  - Port tty.usbmodem142201 Arduino Mega 2560 (Dial-In) opened at 115200 baud, sees DTR: true RTS: true DSR: false CTS: false DCD: false flow: NONE [main]
19:41:46,099 mrix.dccpp.DCCppInitializationManager INFO  - DCC++ Initialization Complete [main]
19:41:46,102 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "Speedo" [main]
19:41:46,109 jmri.jmrix.ConnectionConfigManager    ERROR - Unable to load [Element: <connection/>] into jmri.jmrix.bachrus.kpfserialdriver.configurexml.ConnectionConfigXml [main]
java.lang.IllegalArgumentException: Incorrect stop_bits argument: 8
	at jmri.jmrix.AbstractSerialPortController.activatePort(AbstractSerialPortController.java:185) ~[jmri.jar:?]
	at jmri.jmrix.purejavacomm.CommPortIdentifier.open(CommPortIdentifier.java:22) ~[jmri.jar:?]
	at jmri.jmrix.bachrus.kpfserialdriver.SerialDriverAdapter.openPort(SerialDriverAdapter.java:54) ~[jmri.jar:?]
	at jmri.jmrix.configurexml.AbstractSerialConnectionConfigXml.load(AbstractSerialConnectionConfigXml.java:97) ~[jmri.jar:?]
	at jmri.jmrix.ConnectionConfigManager.initialize(ConnectionConfigManager.java:106) ~[jmri.jar:?]
	at jmri.implementation.JmriConfigurationManager.initializeProvider(JmriConfigurationManager.java:446) ~[jmri.jar:?]
	at jmri.implementation.JmriConfigurationManager.lambda$2(JmriConfigurationManager.java:208) ~[jmri.jar:?]
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:183) [?:?]
	at java.util.stream.SortedOps$SizedRefSortingSink.end(SortedOps.java:357) [?:?]
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:485) [?:?]
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474) [?:?]
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150) [?:?]
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173) [?:?]
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234) [?:?]
	at java.util.stream.ReferencePipeline.forEachOrdered(ReferencePipeline.java:502) [?:?]
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:208) [jmri.jar:?]
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:188) [jmri.jar:?]
	at jmri.implementation.JmriConfigurationManager.load(JmriConfigurationManager.java:178) [jmri.jar:?]
	at apps.AppsBase.setAndLoadPreferenceFile(AppsBase.java:265) [jmri.jar:?]
	at apps.gui3.Apps3.setAndLoadPreferenceFile(Apps3.java:306) [jmri.jar:?]
	at apps.AppsBase.<init>(AppsBase.java:101) [jmri.jar:?]
	at apps.gui3.Apps3.<init>(Apps3.java:73) [jmri.jar:?]
	at apps.gui3.dp3.DecoderPro3.<init>(DecoderPro3.java:25) [jmri.jar:?]
	at apps.gui3.dp3.DecoderPro3.main(DecoderPro3.java:94) [jmri.jar:?]
19:41:46,118 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.jmrix.ConnectionConfigManager: Unable to create connection "Speedo" (A). [main]
19:41:46,335 jmri.jmrit.roster.Roster              INFO  - Reading roster file with rootFromName(/Users/markhendriksen/Library/Preferences/JMRI/Mijn_JMRI_Spoorweg.jmri/roster.xml) [main]
19:41:46,351 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.util.startup.StartupActionsManager: jmri.util.prefs.InitializationException: Door eerdere fouten kunnen de opstartacties niet worden uitgevoerd. [main]
19:41:48,190 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - Station Type set to 'DCC-EX' [AWT-EventQueue-0]
19:41:48,191 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - Build set to 'ee5db61' [AWT-EventQueue-0]
19:41:48,191 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - Version set to '3.2.0' [AWT-EventQueue-0]
19:41:48,210 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - changing maxNumSlots from 12 to 50 [AWT-EventQueue-0]
19:41:50,272 jmri.util.FileUtilSupport             INFO  - File path program: is /Applications/JMRI/ [main]
19:41:50,272 jmri.util.FileUtilSupport             INFO  - File path preference: is /Users/markhendriksen/Library/Preferences/JMRI/Mijn_JMRI_Spoorweg.jmri/ [main]
[... cut ...]
19:41:50,273 jmri.util.FileUtilSupport             INFO  - File path settings: is /Users/markhendriksen/Library/Preferences/JMRI/ [main]
19:41:50,273 jmri.util.FileUtilSupport             INFO  - File path home: is /Users/markhendriksen/ [main]
19:41:50,273 jmri.util.FileUtilSupport             INFO  - File path scripts: is /Applications/JMRI/jython/ [main]
19:42:12,724 jmri.jmrix.bachrus.SpeedoConsoleFrame INFO  - Using Throttle interface for profiling [AWT-EventQueue-0]
19:42:54,760 jmri.jmrix.bachrus.SpeedoConsoleFrame INFO  - Using Throttle interface for profiling [AWT-EventQueue-0]
```

danielb987, 2024-06-05:
> Still does not work.
> DCC++ can connect and be controlled but the speedo is not able to connect.
> In the log is a ERROR.

Thank you! My fault. I have fixed it now. Jenkins will start build shortly.

hendriksen-mark, 2024-06-05:
Yes now it works.
Thanks for the fix and the fast response.

hendriksen-mark, 2024-06-05:
It works on both macos14 and macos10.14

danielb987, 2024-06-05:
> Yes now it works.
> Thanks for the fix and the fast response.

Thank you!

Can you do one more test? I found another bug which is also fixed, but it wasn't included in the Jenkins build you tested. But it is included in the last Jenkins build. The bug happened if JMRI couldn't open the serial port. I just want to make sure nothing is broken.

The Jenkins build is here:
https://builds.jmri.org/jenkins/job/testreleases/job/danielb987/

hendriksen-mark, 2024-06-05:
with the latest i get this with no speed-cat.
also at startup i get a message that the port is not there.
```
23:54:51,983 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]
23:54:52,009 apps.util.Log4JUtil                   INFO  - This log is stored in file: /Users/markhendriksen/Library/Preferences/JMRI/log/session.log [main]
23:54:52,009 apps.util.Log4JUtil                   INFO  - This log is appended to file: /Users/markhendriksen/Library/Preferences/JMRI/log/messages.log [main]
23:54:52,026 apps.AppsBase                         INFO  - DecoderPro version 5.7.7plus+jenkins+20240605T2058Z+Rff1e80d380 starts under Java 11.0.23 on Mac OS X x86_64 v14.5 at Wed Jun 05 23:54:52 CEST 2024 [main]
23:54:52,187 apps.gui3.Apps3                       INFO  - Starting with profile Mijn_JMRI_Spoorweg.3f010714 [main]
23:54:53,465 jmri.util.node.NodeIdentity           INFO  - Using e0343de8-53a7-40bf-aa0c-35837e8357bb as the JMRI storage identity for profile id 3f010714 [AWT-EventQueue-0]
23:54:53,531 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "DCC++" [main]
23:54:53,726 jmri.jmrix.dccpp.serial.DCCppAdapter  INFO  - Connecting DCC++ to cu.usbmodem142201 Arduino Mega 2560 [main]
23:54:53,729 jmri.jmrix.dccpp.serial.DCCppAdapter  INFO  - Port cu.usbmodem142201 Arduino Mega 2560 opened at 115200 baud, sees DTR: true RTS: true DSR: false CTS: false DCD: false flow: NONE [main]
23:54:53,792 mrix.dccpp.DCCppInitializationManager INFO  - DCC++ Initialization Complete [main]
23:54:53,796 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "Speedo" [main]
23:54:53,805 jmrix.purejavacomm.CommPortIdentifier ERROR - Serial port cu.usbmodem142301 not found: This port appears to have been shutdown or disconnected. [main]
23:54:53,807 s.kpfserialdriver.SerialDriverAdapter ERROR - Serial port cu.usbmodem142301 not found [main]
23:54:53,807 jmri.jmrix.ConnectionConfigManager    ERROR - Unable to create jmri.jmrix.bachrus.kpfserialdriver.configurexml.ConnectionConfigXml for [Element: <connection/>], load returned false [main]
23:54:53,808 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.jmrix.ConnectionConfigManager: Unable to create several of your connections. [main]
23:54:54,163 jmri.jmrit.roster.Roster              INFO  - Reading roster file with rootFromName(/Users/markhendriksen/Library/Preferences/JMRI/Mijn_JMRI_Spoorweg.jmri/roster.xml) [main]
23:54:54,181 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.util.startup.StartupActionsManager: jmri.util.prefs.InitializationException: Door eerdere fouten kunnen de opstartacties niet worden uitgevoerd. [main]
23:54:55,858 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - Station Type set to 'DCC-EX' [AWT-EventQueue-0]
23:54:55,858 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - Build set to 'ee5db61' [AWT-EventQueue-0]
23:54:55,858 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - Version set to '3.2.0' [AWT-EventQueue-0]
23:54:55,878 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - changing maxNumSlots from 12 to 50 [AWT-EventQueue-0]
23:55:02,562 jmri.util.FileUtilSupport             INFO  - File path program: is /Applications/JMRI/ [main]
23:55:02,563 jmri.util.FileUtilSupport             INFO  - File path preference: is /Users/markhendriksen/Library/Preferences/JMRI/Mijn_JMRI_Spoorweg.jmri/ [main]
23:55:02,563 jmri.util.FileUtilSupport             INFO  - File path profile: is /Users/markhendriksen/Library/Preferences/JMRI/Mijn_JMRI_Spoorweg.jmri/ [main]
23:55:02,563 jmri.util.FileUtilSupport             INFO  - File path settings: is /Users/markhendriksen/Library/Preferences/JMRI/ [main]
23:55:02,563 jmri.util.FileUtilSupport             INFO  - File path home: is /Users/markhendriksen/ [main]
23:55:02,563 jmri.util.FileUtilSupport             INFO  - File path scripts: is /Applications/JMRI/jython/ [main]
```

this is with speed-cat connected.
```
23:57:28,393 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]
23:57:28,417 apps.util.Log4JUtil                   INFO  - This log is stored in file: /Users/markhendriksen/Library/Preferences/JMRI/log/session.log [main]
23:57:28,417 apps.util.Log4JUtil                   INFO  - This log is appended to file: /Users/markhendriksen/Library/Preferences/JMRI/log/messages.log [main]
23:57:28,429 apps.AppsBase                         INFO  - DecoderPro version 5.7.7plus+jenkins+20240605T2058Z+Rff1e80d380 starts under Java 11.0.23 on Mac OS X x86_64 v14.5 at Wed Jun 05 23:57:28 CEST 2024 [main]
23:57:28,597 apps.gui3.Apps3                       INFO  - Starting with profile Mijn_JMRI_Spoorweg.3f010714 [main]
23:57:29,928 jmri.util.node.NodeIdentity           INFO  - Using e0343de8-53a7-40bf-aa0c-35837e8357bb as the JMRI storage identity for profile id 3f010714 [AWT-EventQueue-0]
23:57:29,989 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "DCC++" [main]
23:57:30,043 jmri.jmrix.dccpp.serial.DCCppAdapter  INFO  - Connecting DCC++ to cu.usbmodem142201 Arduino Mega 2560 [main]
23:57:30,046 jmri.jmrix.dccpp.serial.DCCppAdapter  INFO  - Port cu.usbmodem142201 Arduino Mega 2560 opened at 115200 baud, sees DTR: true RTS: true DSR: false CTS: false DCD: false flow: NONE [main]
23:57:30,092 mrix.dccpp.DCCppInitializationManager INFO  - DCC++ Initialization Complete [main]
23:57:30,096 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "Speedo" [main]
[... cut ...]
23:57:32,172 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - Station Type set to 'DCC-EX' [AWT-EventQueue-0]
23:57:32,172 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - Build set to 'ee5db61' [AWT-EventQueue-0]
23:57:32,172 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - Version set to '3.2.0' [AWT-EventQueue-0]
23:57:32,192 jmri.jmrix.dccpp.DCCppCommandStation  INFO  - changing maxNumSlots from 12 to 50 [AWT-EventQueue-0]
23:57:38,321 jmri.jmrix.bachrus.SpeedoConsoleFrame INFO  - Using Throttle interface for profiling [AWT-EventQueue-0]
```

danielb987, 2024-06-05:
> with the latest i get this with no speed-cat.
> also at startup i get a message that the port is not there.

Thank you! It looks good.

I have merged PR #13163 so it will be included in the next JMRI test release.

closed 2024-06-10 by hendriksen-mark