JMRI/JMRI #13177: JMRI will not launch under MAC Sonoma 14.5

jayfmn, 2024-06-06:
Hi, I have tried the production version & beta version. Both set the icon in the dock & that’s it.
There are no messages relating to JMRI in the Macs log.
This happens with the other 2 apps also. 
I am stumped!

dsand47, 2024-06-06:
Check the logs in the "/Users/<username>/Library/Preferences/JMRI/log" directory.

Look at the "launcher.log" file first.  Open it with TextEdit, copy and paste the contents in a reply.  Do the same for the session.log file

Dave Sand

jayfmn, 2024-06-06:
Sorry Dave, I can find no such file. I came from osx 11.7. 14.5 is a totally different animal.

dsand47, 2024-06-07:
How are doing the JMRI install?

The process is to download the DMG from the JMRI web site.  Double click on the DMG to open it.  There will be two icons, one for JMRI and a shortcut to /Applications.  Drag the JMRI icon to the Applications icon.  If you get a prompt about merge or replace, always select replace.

Is this an initial JMRI install or are you upgrading?

Dave Sand

jayfmn, 2024-06-07:
Hi Dave, this is an initial install. I tried the production version, didn’t work. I then went to the beta version, did a replace. I then downloaded Zulu jre 11.0.23, placed in the Jeri folder & renamed it jre per the jmri website.

jayfmn, 2024-06-07:
Auto correct on my phone is at least annoying! Jeri folder is jmri!!

dsand47, 2024-06-07:
Is your Mac an Intel or M series?

jayfmn, 2024-06-07:
It’s an M3 Mac. Latest & greatest :-)

dsand47, 2024-06-08:
I recommend that you install the ARM version of Azul Java.

Download the .dmg package.  Double click on the .dmg to open it.  Double click on the installer icon.

When the Java install is done, it will be located in the system library.
[image]

You only need to use the alternate "jre" approach is when you have other software that requires a different version of Java.  This approach also requires repeating the jre process with each JMRI upgrade.

I recommend that you use JMRI 5.7.7 test release.

Download the JMRI dmg and copy the JMRI icon to the Applications icon.  This step is important.

The run PanelPro first.  The first run will have the typical application was downloaded from the internet warning.  After that has been acknowledged, PanelPro will need you to create a Connection.  I recommend you use a simulated connection for your command station to get started.

jayfmn, 2024-06-08:
I did a fresh install from scratch, it now works? Idk, but it did launch. Thank you for your time!

closed 2024-06-21 by bobjacobsen