JMRI/JMRI #13179: Request for an additional decoder definition: NONE

asbrack, 2024-06-08:
Issue Type: __Feature Request__

Could you add a decoder definition for no decoder (ID 0) so that we can keep track of locomotives that do not have a decoder installed?

DecoderPro 5.7.4+Rb028ee0124
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 11.0.21+9-LTS-193 mixed mode
Windows 10 10.0 amd64

bobjacobsen, 2024-06-08:
@asbrack Interesting idea!  Can you help me think through some of the details?

[image]

First, how would you like it to appear in the selection tree?  JMRI uses a manufacturer/model approach to that.  What should we call those?  Perhaps "None" and "No Decoder"? Or "Analog Loco"? Or maybe "DC"?  Would appreciate your ideas on how to do this best.

Second, if we just add that the selection tree in alphabetical order, will people even know it's there?  Maybe we should move it to the top, or perhaps right under the NMRA definitions at that top?  But is it that important?

Finally, DecoderPro is built around the idea of roster entries carrying an address, which is then used as part of the selection.  So we'll need to give these entries a decoder address as a number.  It can't just be blank. Does zero make sense for that?

Thanks in advance for your thoughts.  And thanks for the suggestion!

danielb987, 2024-06-08:
> First, how would you like it to appear in the selection tree? JMRI uses a manufacturer/model approach to that. What should we call those? Perhaps "None" and "No Decoder"? Or "Analog Loco"? Or maybe "DC"?

I vote for "No decoder".
Analog Loco or DC is not appropriate since they focus on DCC. But some people run trains which are controlled by radio with battery in the loco, for example garden railways.

> Second, if we just add that the selection tree in alphabetical order, will people even know it's there? Maybe we should move it to the top, or perhaps right under the NMRA definitions at that top? But is it that important?

I vote for having it at the top. It's a complete different kind of beast than the others. And having it at the top helps people finding it and knowing it's there.

> Finally, DecoderPro is built around the idea of roster entries carrying an address, which is then used as part of the selection. So we'll need to give these entries a decoder address as a number. It can't just be blank. Does zero make sense for that?

My suggestion is that it has an address with the default "0", but user selectable. Some DCC systems allow running analog locos using address 0. And there are ways of running more than one analog loco at once, for example one loco using catenary and one loco using track. Or running multiple trains without decoder but radio controlled. JMRI doesn't natively supports that currently, but it can be implemented using Jython scripts or using external hardware like MQTT.

asbrack, 2024-06-08:
I think something like “<none>”, “- none -“ or “*none*” or maybe just “<no decoder>” would be appropriate. The reason for suggesting the punctuation as a part of the name is that it both stands out and will likely never be the name of a manufacturer’s decoder. This also moves it to the top of the sort order.

 

Address 0 makes sense for a number of reasons, not the least of which is that Digitrax uses address 0 for running DC locomotives, and the fact that 0 is not an otherwise valid address. To my knowledge, you cannot program a decoder to actually have address 0. Alternatively if you must have an alternative, -1 or 0xFFFF could be used as well as I do not believe that is a valid decoder address since decoder addresses are expressed as a 2 or 4 digit decimal number.

(see below from the WiKi)

 

Regarding your comment of running more than 1 loco on address 0, the same holds true for any other address in JMRI’s roster. The only thing that needs to be unique (and is) is the name within the roster. The key thing, again, is that I do not believe that 0 is an allowable decoder address, but we should review the DCC standard to make sure.

closed 2024-06-08 by bobjacobsen