JMRI/JMRI #13186: Atlas N scale decoder #522299 (MP15DC) not listed

asbrack, 2024-06-10:
Issue Type: __Feature Request__

This is just an FYI that the decoder is not listed. It seems to be similar to the decoder for a VO1000, so that is what I have used instead.

Tony

DecoderPro 5.7.4+Rb028ee0124
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 11.0.21+9-LTS-193 mixed mode
Windows 10 10.0 amd64

bobjacobsen, 2024-06-10:
Thanks!  When you have a moment, could you read the value of CV7 and CV8 from that decoder and post them here?  Thanks again.

asbrack, 2024-06-10:
Hi Bob,

 

CV 7 = 45

CV 8 = 127 (Lenz)

 

It is the same numbers as the Atlas (Lenz) decoder for a GP 7/9 that also isn’t listed, but that I do not have a part number for. If I find a part number for that one, I will let you know as well.

PUlvestad, 2024-06-11:
According to the Atlas website http://download.atlasrr.com/pdf/Atlas_N-Scale_Decoder_Applications.pdf
you would use the LE063XF definition which is listed. It should have been one of the options when using the "New Loco".

Peter

asbrack, 2024-06-12:
An LE063XF is the standard short decoder. It will not fit a GP7/9/35 for sure, so that chart is incorrect. It may, however, give us the correct Atlas part number: 508299. That Lenz decoder is the standard Atlas short decoder, which I am positive will not fit.

 

Digitrax uses DN163A4, versus the DN163A0 which is the standard short decoder and is the same size as the LE063XF which fits most engines.

 

Given what I know about the physical size, that page is wrong, although they are logically the same and probably use the same chip (as does LE062XF, the long SD60 decoder).

PUlvestad, 2024-06-12:
The sheet lists 2 different decoders for the GP 7/9/35 depending on vintage.

asbrack, 2024-06-12:
That’s nice, but LE063XF is physically about ½”  too long to fit either version of the locomotive even though they are likely the same electronically. I am not at home, but I can re-read the decoder to see how it identifies itself. Regardless, LE063XF is a standard short Atlas decoder that fits most of their models, but is not the same physical form factor (length) as the GP7/9/30/35 decoders.

If I recall, it identified itself identically to the way MP-15DC decoder I provided did.

 

When I get home, I will re-read the actual decoder to confirm how it identifies itself, and maybe pop the shell to see if there is a visible part number. Most of mine came from Atlas with either Digitrax or TCS decoders, but this one showed up looking like the MP15 decoder I identified, and based on the Digitrax DN163A3 (MP15) and Digitrax DN163A4 (GP9) … those 2 are very close in physical size.

asbrack, 2024-06-14:
CV7 = 45

CV8 = 127

bobjacobsen, 2024-06-14:
See PR #13189

closed 2024-06-16 by bobjacobsen