JMRI/JMRI #13190: DecoderPro needs to recognize the ESU LokPilot Nano DCC 59925 (NON-sound) decoder

rick77337, 2024-06-15:
Atlas is flooding the market with their new (Jan 2024) N scale GP7 which can **only** take the new ESU 59925 24 pin decoder, so this is needed quickly.
(I am assuming their soon to arrive RS-3 will also need it.)

danielb987, 2024-06-15:
@rick77337 
Welcome to the JMRI project!

@bobjacobsen 
Is this something for you to take a look at?

bobjacobsen, 2024-06-15:
@rick77337 Do you have one of these decoders?  If so, could you send me the identification information?  You can get this using the "Single CV Programmer" from the "Programmers" section of the "Tools" menu. The product ID CVs are indexed, so you have to write some values to the index.  That won't cause any problems or make any changes to your decoder.  The sequence is:

Read CV7
Read CV8

Write 0 to CV31
Write 255 to CV32

Read CV261
Read CV262
Read CV263
Read CV264

If you could provide the results of those read operations, that would be very helpful!  Thank you in advance.

rick77337, 2024-06-24:
Haven't forgotten your generous offer of help. Have gotten myself locked out of the account and now getting a 306 error with Decoder Pro (was working fine for months). Will be back as soon as possible. Thanks

rick77337, 2024-06-24:
CV7: 255
CV8: 151
Write 0 to CV31 (previously was 
Read16)
Write 255 to CV32 (previously was 6)
Read CV261: 252
Read CV262: 0
Read CV263: 0
Read CV264: 1
The problem may be the initial JMRI decoder selection. Since there was nothing for the LocPilot 5 Nano DCC 24 pin, JMRI attempting to read the decoder just highlightled almost all the esu decoders to choose from and I believe I picked the LokPilot 5 Micro DCC (18 pin?) which seemed the closest. I used a speed table (log & shifted left). The actual speed control takes forever to create speed and I have no lights.

rick77337, 2024-06-28:
I finally got hold of ESU tech support and they suggested using the "LokPilot 5 Micro DCC" framework on JMRI it did allow me to program the decoder road# and use the speed table with acceptable results.
(The high speed limit is kinda low).

A lot of the higher CV responses were not acknowledged (responded to) returning 308 error codes (no acknowledgement).

When I programmed the speed table the first time I got red (unacknowledged) first and last bars but all the other intermediate bars acknowledged the changes.
When I attempted to redo the speed table all responses failed acknowledgement.

What has failed is headlight control.
All I am looking for is the headlight going on (medium to bright) in the direction of engine travel only.

In my 3 identical units:
1. No lights at all
2. Cab light only, no matter the direction of travel
3. Long hood light only, no matter the direction of travel

I do NOT have on/off light control on any of the units.

rick77337, 2024-07-04:
From: Richard Balassaitis <rick773@hotmail.com>
Sent: Thursday, July 4, 2024 2:49 PM
To: Lee Ryan <Lee.Ryan@esu.eu>
Subject: Re: ESU 59925 (Atlas N Scale NON-sound GP7/9)
Lee, 
I hope you enjoyed a pleasant holiday.
I have tried to enter the CVs you suggested most were rejected and the light problem continued.
I took one of my 3 engines and reprogrammed it using LokPilot 5 DCC (per your suggeation) and the exact same conditions resulted as with the LokPilot 5 Micro  DCC.
I reseated the decoder with same results.
The programming track has been cleaned (again) and we are dealing with 3 identical engine frames and 59925 decoders so I don't think track pick-up or /motor connections are to blame.
I think Atlas made a mistake using this decoder on their new GP7s unless ESU wants to add a free LokProgrammer ($144-$180) with each decoder or pay their dealers to program the lights.
Let me know when ESU provides you with an Atlas GP7 (40 005 341 class) and 59925  decoder for your testing.
I am putting all your suggestions on the JMRI user forum. So far no one else has provided any suggestions so I guess there is currently no known fix.
Thanks for all your assistance,
Looking forward to hearing from you again,
Rick

From: Lee Ryan <Lee.Ryan@esu.eu>
Sent: Friday, June 28, 2024 1:26 PM
Hi Richard,
One thing to note is that ESU decoders don’t use CV67 for speed step 1 and CV94 for speed step 28 on the curve. Speed step 1 is handled by CV2 and Speed step 28 is handled by CV5. This is noted below the speed table in Decoder Pro.
I am surprised that reading the speed table that those would fail acknowledgement. That really makes no sense. Those CVs are standardized by the NMRA.
You might be able to get a little more speed out of the locomotive by increasing CV53. It may be set too low, I would try a value of 130 or 140. This is the BEMF Reference Voltage on the Motor tab in JMRI.
I am also a bit surprised that the higher CVs wouldn’t work either. I hate to say it but this seems like a connection issue with the locomotive somewhere. It seems that either the track pick-up or the connection to the motor is spotty. Like it’s making partial contact.
The other mystery here is that by default the headlights/rear lights are programmed to work out of the box with no extra programming needed.
I also wonder if the better decoder to choose would be the LokPilot 5 DCC rather than the LokPilot 5 Micro DCC.
By default, “F0, Forward” is the front light and “F0, Reverse” is the rear light. The CVs for these are by default:
CV31 = 16 FIRST (default) may not really need to be programmed
CV32 = 3 SECOND
CV257 = 20 (in JMRI this would be 16.3.257 = 20)
CV273 = 24 (in JMRI this would be 16.3.273 = 24)
Then set:
CV32 = 8 FIRST
CV257 = 1 (in JMRI this would be 16.8.257 = 1)
CV273 = 2 (in JMRI this would be 16.8.273 = 2)
If you wanted to add in the Rule 17 to make them go dim when stopped and bright when moving with the fade in/out effect as well as LED mode (even though LED mode isn’t totally necessary):
CV31 = 16 FIRST (default) may not really need to be programmed
CV32 = 0 SECOND
CV259 = 2 (in JMRI this would be 16.0.259 = 2)
CV263 = 132 (in JMRI this would be 16.0.263 = 132)
CV267 = 2 (in JMRI this would be 16.8.267 = 2)
CV271 = 136 (in JMRI this would be 16.8.271 = 136)
If the lights still are not working properly, you might check all of the electrical connections as well as re-seat the decoder and make sure it is snapped down all the way, the E24 socket can be deceiving and the decoder can be angled.
I have done some testing with JMRI with a different LokPilot decoder matching all of the same parameters, and all of this is working as expected. Admittedly I don’t have a LokPilot 5 Nano on hand (I work in a remote office in Colorado, not the main office in PA) nor an Atlas GP7/9. I may see if I can get these items though so I can fully test this but I don’t see any reason why it wouldn’t work.
Best Regards,
Lee Ryan

rick77337, 2024-07-11:
For the FAILED conclusion of this topic see:
https://www.therailwire.net/forum/index.php?topic=58119.0
Until someone can come up with a JMRI fix for the new ESU LokPilot 5 NANO DCC 24 pin. Atlas Silver Series GP7/9 owners are screwed.

PR #13402 (2024-09-03): ESU decoder definition update
- Added LokPilot 5 micro DCC Direct, LokPilot 5 nano DCC, LokSound 5 micro DCC Direct Atlas S2 and LokSound 5 nano DCC Next18 (untested, don't have the hardware)
- Added a few productIDs to LokPilot 5 L and LokPilot 5 L DCC
- Moved LokSound 5 micro DCC Direct Atlas Legacy definition to match LokProgrammer decoder order
- Removed SUSI function configuration from LokSound 5 micro DCC Direct Atlas Legacy

Should fix #13190 from the JMRI decoder definition side. I believe there's more to rick77337's troubles than just a missing decoder definition, however...

With thi [... cut]
commit 47e8d24f (2024-09-03): ESU decoder definition update

- Added LokPilot 5 micro DCC Direct, LokPilot 5 nano DCC, LokSound 5 micro DCC Direct Atlas S2 and LokSound 5 nano DCC Next18 (untested, don't have the hardware)
- Added a few productIDs to LokPilot 5 L and LokPilot 5 L DCC
- Moved LokSound 5 micro DCC Direct Atlas Legacy definition to match
  LokProgrammer decoder order
- Removed SUSI function configuration from LokSound 5 micro DCC Direct
  Atlas Legacy

Should fix #13190 from the JMRI decoder definition side. I believe
there's more to rick77337's troubles than just a missing decoder
definition, however...

Wit [... cut]
closed 2024-09-03 by danielb987