JMRI/JMRI #13192: Changes to LoxigNG "Enabled" does not trigger panel store dialog on JMRI shutdown

jerryg2003, 2024-06-15:
If a LogixNG "enabled" flag is checked or unchecked during a session, but the user has not stored the panel file, JMRI shutdown does not cause a warning window popup like most other changes to panel information.  This means that the user's action of enabling/disabling LogixNGs will be lost in this case (it is saved properly if the user remembers to store the panel file before exiting JMRI). 

Suggest that LogixNG enabling flag changes should trigger the "you haven't stored your panel file" warning dialog before JMRI shutdown.

dsand47, 2024-06-15:
It works for me.

```
16:07:00,435 jmri.configurexml.StoreAndCompare     INFO  - Match failed in StoreAndCompare: [AWT-EventQueue-0]
16:07:00,436 jmri.configurexml.StoreAndCompare     INFO  -     file1:line 33: "    <LogixNG enabled="yes" inline="no">" [AWT-EventQueue-0]
16:07:00,436 jmri.configurexml.StoreAndCompare     INFO  -     file2:line 33: "    <LogixNG enabled="no" inline="no">" [AWT-EventQueue-0]
16:07:00,436 jmri.configurexml.StoreAndCompare     INFO  -   comparing file1:"/Users/das/JMRI/_Profiles/LogixNG.jmri/execute-action-test.xml" [AWT-EventQueue-0]
16:07:00,436 jmri.configurexml.StoreAndCompare     INFO  -          to file2:"/var/folders/fb/zsjrbj816fx9x2kzxznnrgmr0000gp/T/5f8c4790-04ba-43ca-a3b4-40341983cf53.xml" [AWT-EventQueue-0]
```

The **Debug &rArr; Test Shutdown Store and Compare**  menu item is handy for previewing the shutdown notification process.

jerryg2003, 2024-06-16:
Thanks @dsand47.  Didn't know about that debugging tool.  And, of course, now I can't replicate the issue, even though I did replicate it (3 times) before submitting it...sigh.

closed 2024-06-16 by jerryg2003