JMRI/JMRI #13199: System Crash when printing Turnout sheet

RHinebaugh, 2024-06-17:
Issue Type: __Bug__

Open turnout table and select print.

PanelPro 5.7.9+R80852804ee
OpenJDK 64-Bit Server VM OpenLogic 21.0.3+9-adhoc.Administrator.jdk21u mixed mode
Windows 10 10.0 amd64

[session.log](https://github.com/user-attachments/files/15873409/session.log)
[systemInfo.txt](https://github.com/user-attachments/files/15873410/systemInfo.txt)

bobjacobsen, 2024-06-17:
Thanks for providing the log.  Unfortunately, it doesn't show any information to help us track this down.  What does the crash look like?  Is JMRI just closing, or are you seeing something else?

RHinebaugh, 2024-06-17:
Here is a clipboard COPY.

Unrecoverable error encountered.
This application will quit.

HTHEME is null
java.lang.InternalError
java.lang.InternalError: HTHEME is null
at java.desktop/sun.awt.windows.ThemeReader.paintBackground(Native Method)
at java.desktop/sun.awt.windows.ThemeReader.paintBackground(Unknown Source)
at java.desktop/com.sun.java.swing.plaf.windows.XPStyle$SkinPainter.paintToImage(Unknown Source)
at java.desktop/sun.swing.CachedPainter.getImage(Unknown Source)
at java.desktop/sun.swing.CachedPainter$PainterMultiResolutionCachedImage.getResolutionVariant(Unknown Source)
at java.desktop/sun.java2d.SunGraphics2D.getResolutionVariant(Unknown Source)
at java.desktop/sun.java2d.SunGraphics2D.drawHiDPIImage(Unknown Source)
at java.desktop/sun.java2d.SunGraphics2D.drawImage(Unknown Source)
at java.desktop/sun.java2d.SunGraphics2D.drawImage(Unknown Source)
at java.desktop/sun.print.PeekGraphics.drawImage(Unknown Source)
at java.desktop/sun.swing.CachedPainter.paintImage(Unknown Source)
at java.desktop/sun.swing.CachedPainter.paint0(Unknown Source)
at java.desktop/sun.swing.CachedPainter.paint(Unknown Source)
at java.desktop/com.sun.java.swing.plaf.windows.XPStyle$Skin.paintSkinRaw(Unknown Source)
at java.desktop/com.sun.java.swing.plaf.windows.AnimationController.paintSkin(Unknown Source)
at java.desktop/com.sun.java.swing.plaf.windows.XPStyle$Skin.paintSkin(Unknown Source)
at java.desktop/com.sun.java.swing.plaf.windows.WindowsTableHeaderUI$XPDefaultRenderer.paint(Unknown Source)
at java.desktop/javax.swing.CellRendererPane.paintComponent(Unknown Source)
at java.desktop/javax.swing.plaf.basic.BasicTableHeaderUI.paintCell(Unknown Source)
at java.desktop/javax.swing.plaf.basic.BasicTableHeaderUI.paint(Unknown Source)
at java.desktop/javax.swing.plaf.ComponentUI.update(Unknown Source)
at java.desktop/javax.swing.JComponent.paintComponent(Unknown Source)
at java.desktop/javax.swing.JComponent.printComponent(Unknown Source)
at java.desktop/javax.swing.JComponent.paint(Unknown Source)
at java.desktop/javax.swing.JComponent.print(Unknown Source)
at java.desktop/javax.swing.TablePrintable.print(Unknown Source)
at java.desktop/javax.swing.JTable$ThreadSafePrintable$1.run(Unknown Source)
at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
at java.base/java.security.AccessController.doPrivileged(Unknown Source)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
at java.desktop/java.awt.WaitDispatchSupport$2.run(Unknown Source)
at java.desktop/java.awt.WaitDispatchSupport$4.run(Unknown Source)
at java.desktop/java.awt.WaitDispatchSupport$4.run(Unknown Source)
at java.base/java.security.AccessController.doPrivileged(Unknown Source)
at java.desktop/java.awt.WaitDispatchSupport.enter(Unknown Source)
at java.desktop/java.awt.Dialog.show(Unknown Source)
at java.desktop/java.awt.Component.show(Unknown Source)
at java.desktop/java.awt.Component.setVisible(Unknown Source)
at java.desktop/java.awt.Window.setVisible(Unknown Source)
at java.desktop/java.awt.Dialog.setVisible(Unknown Source)
at java.desktop/sun.swing.PrintingStatus.showModalOnEDT(Unknown Source)
at java.desktop/sun.swing.PrintingStatus.showModal(Unknown Source)
at java.desktop/javax.swing.JTable.print(Unknown Source)
at java.desktop/javax.swing.JTable.print(Unknown Source)
at java.desktop/javax.swing.JTable.print(Unknown Source)
at jmri.jmrit.beantable.AbstractTableTabAction.print(AbstractTableTabAction.java:128)
at jmri.jmrit.beantable.ListedTableFrame.lambda$buildMenus$2(ListedTableFrame.java:270)
at java.desktop/javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
at java.desktop/javax.swing.DefaultButtonModel.setPressed(Unknown Source)
at java.desktop/javax.swing.AbstractButton.doClick(Unknown Source)
at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
[... 81 lines cut ...]
user.timezone: America/Phoenix
jmri.log.path: C:\Users\Rick\JMRI\log\
FileSystemView#getDefaultDirectory(): C:\Users\Rick\OneDrive\Documents
FileSystemView#getHomeDirectory(): C:\Users\Rick\Desktop
Default JFileChooser(): C:\Users\Rick\OneDrive\Documents
Screen size h:768, w:1366 Inset t:0, b:0; l:0, r:0
Screen resolution: 96
Environment max bounds: java.awt.Rectangle[x=0,y=0,width=1366,height=768]
Device: \Display0 bounds = java.awt.Rectangle[x=0,y=0,width=1366,height=768]
Device: \Display0 insets = java.awt.Insets[top=0,left=0,bottom=0,right=0]
JmriInsets t:0, b:0; l:0, r:0
Look and Feel: Windows
Network Interface: ethernet_32772
Long Name: Intel(R) Ethernet Connection I217-LM
Host Name: TRAINS
IP address: 192.168.1.133
[... cut ...]
ZeroConfService host: jmri-c4346b5336cd-3e4d06fa-3 running 0 service(s)
ZeroConfService host: jmri-c4346b5336cd-3e4d06fa-3 running 0 service(s)
ZeroConfService host: jmri-c4346b5336cd-3e4d06fa-3 running 0 service(s)
ZeroConfService host: jmri-c4346b5336cd-3e4d06fa-1 running 0 service(s)
ZeroConfService host: jmri-c4346b5336cd-3e4d06fa-3 running 0 service(s)
ZeroConfService host: jmri-c4346b5336cd-3e4d06fa running 0 service(s)
ZeroConfService host: jmri-c4346b5336cd-3e4d06fa-3 running 0 service(s)

bobjacobsen, 2024-06-17:
Thanks.

There's a discussion of what seems to be a similar bug on the [jmri users list](https://groups.io/g/jmriusers/topic/106449578#msg231250).  No resolution there, unfortunately.

danielb987, 2024-06-17:
@RHinebaugh 
If you can, try to change the Look&Feel from Windows to Metal in JMRI.

RHinebaugh, 2024-06-17:
Changed Look&Feel to Metal worked.  No crash when printing.   Thanks

nschertz, 2024-07-01:
Sorry.. Steve Young not Steve G.

closed 2024-09-23 by github-actions[bot]