JMRI/JMRI #13218: Create Production Release 5.8

bobjacobsen, 2024-06-24:
[image]

bobjacobsen, 2024-06-24:
JMRI 5.8 has been released. Files are available in the GitHub release section.

Sammybigue, 2024-07-01:
> ![image](https://private-user-images.githubusercontent.com/2774117/342326224-2828103e-60ca-4b3c-b8b0-fd7fc2f85c89.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk4NDIxMjksIm5iZiI6MTcxOTg0MTgyOSwicGF0aCI6Ii8yNzc0MTE3LzM0MjMyNjIyNC0yODI4MTAzZS02MGNhLTRiM2MtYjhiMC1mZDdmYzJmODVjODkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDFUMTM1MDI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NGM4NGY0OWI1OTgxNWNkNjdjYTY5MWIwNWY2ZDg0OTlmZmZiM2Y5Y2U1NTliMDM4ZWUwNGM2YzJmZDJjNmE0MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Til-AfXgY_DkCe_QbRqxYTyeaY-OYIzHfAcdWG-QrEk)

closed 2024-07-12 by bobjacobsen