JMRI/JMRI #13244: TMCC Horn only continues to play if F2 is consistently pressed

tomixnscale89, 2024-07-09:
Hello,

JMRI Version: JMRI 5.8+Rbc21ce2ce7
Windows 11

While using JMRI 5.8+Rbc21ce2ce7 (and older versions) with TMCC, I noticed that pressing the F2 function key will not continue to send horn commands even when the "Latchable" tickbox is enabled. The only way to have the horn consistently play is to press the F2 button repeatedly.

In order for TMCC trains to continually play their horn, they need to have the horn command sent out consistently.

bobjacobsen, 2024-07-09:
Is this true of other functions also?  Or is it just F2?

tomixnscale89, 2024-07-09:
Someone from Lionel said 100ms for the horn should be enough. Think we should give that a shot.

https://ogrforum.ogaugerr.com/topic/77875232601700913?reply=80832753551815751#80832753551815751

bobjacobsen, 2024-07-09:
When you get the chance, could you try one of the installers from this page:

https://builds.jmri.org/jenkins/job/testreleases/job/bobjacobsen/

These have a modification to resend each function command so long as the function stays active.  If it's working OK for you, I'll get this merged to the main code.

tomixnscale89, 2024-07-10:
Thank you Bob for making these changes, I really appreciate it. It works great now. 

I have some questions about more Lionel Legacy functionality, but I'll create a different issue for this.

closed 2024-07-11 by bobjacobsen