JMRI/JMRI #13248: Lionel Legacy Speed Step Implementation

tomixnscale89, 2024-07-10:
Lionel Legacy trains can use the existing Lionel TMCC JMRI supports, but the TMCC command set only is capable of 32 speed steps. The Legacy protocol, supports up to 200 speed steps.

Is there a way that this change in speed step could be implemented? Such as JMRI reading a particular decoder definition (say Legacy Steam vs TMCC Steam) to know which type of command to send. 

You can find the Legacy speed step command on page eight here:
https://ogrforum.ogaugerr.com/fileSendAction/fcType/0/fcOid/156701992259624343/filePointer/156701992265497355/fodoid/156701992265497351/LCS-LEGACY-Protocol-Spec-v1.21.pdf

bobjacobsen, 2024-07-10:
I'm not sure it's practical to add this to a roster entry.

With other types of locomotives, you can set the speed steps used by a throttle by ctrl-clicking (right clicking) on the top of the speed sub-window.  That gives you a combo box to select 14, 28 and 128 speed steps for DCC, etc. Right now, for TMCC locomotives the only choice is "32 SS (TMCC)".

We could maybe add "200 SS (TMCC Legacy)" to that.  I'll have to look a little further.

bobjacobsen, 2024-07-10:
Could you try the installers at:

https://builds.jmri.org/jenkins/job/testreleases/job/bobjacobsen/127/

and let me know if this works?  The TMCC2 200 steps choice should be available above the speed slider on the JMRI throttle.

tomixnscale89, 2024-07-11:
Bob, 

The commands are working great, thank you. I noticed two things:

1. At speed step 200, you send the command for "Set Momentum: 0". The speed step command field data should be from 0-199, so I think you're going one over. For example, at speed step 200, with engine address 20, I see:

F8 34 C8

Looking at the command bits shows you are sending 0 1100 1000, which is actually for the momentum.  If you go one decimal less, then it would be correct:

F8 34 C7

0 11 00 0111

2. The serial line gets bogged down quite a lot when issuing speed commands with the middle mouse wheel. Initially I thought the speed commands were being slowly transitioned, but noticed this behavior is only for the mouse scroll wheel. 

I have some more thoughts on fine tuning/adding more Legacy commands. Should I create a separate issue for this?

bobjacobsen, 2024-07-11:
There are new installers at https://builds.jmri.org/jenkins/job/testreleases/job/bobjacobsen/129/ that should (1) top out at 199 instead of 200 and (2) handle the middle mouse wheel _a_ _little_ better.  

It would be great if you could create another issue with further suggestions.  No promises on implementation, though.

tomixnscale89, 2024-07-12:
Bob,

The limit works good now, thank you. 

I tried the scroll wheel fix, it does help, but I feel like the code doesn't need to ramp up/down the speed for TMCC/Legacy locomotives. With these engines, it's the locomotive's responsibility to slow down or speed up (with the momentum buttons). If that change delves too deep into the core JMRI code, I think it could be left for now.

tomixnscale89, 2024-07-12:
Another thing I noticed, EngineDriver defaults to using 32 TMCC speed steps unless I open the loco first with JMRI and change the speed steps to 200. Is there a workaround for this?

bobjacobsen, 2024-07-12:
Thanks for testing this. It'll be in the next test release.

The thumbwheel is a tricky thing.  The probably cause is that your thumbwheel is sending mouse-movement events, each of which JMRI sees as an independent request to set the speed.  I'll look into it a little more.

I don't know much about Engine Driver.  You might want to ask on the JMRIusers list about that.

closed 2024-07-13 by danielb987