JMRI/JMRI #13253: Improving Lionel Legacy Functions

tomixnscale89, 2024-07-12:
I've been adding some more legacy functions on my local branch of JMRI. I added the following functions:

        0xF8013B, // Fn32 Aux 3
        0xF8012D, // Fn33 Locomotive Re-Fueling Sound
        0xF801F8, // Fn34 RS Trigger Brake Air-Release
        0xF801F9, // Fn35 RS Trigger, Short Let-Off
        0xF801FA, // Fn36 RS Trigger, Long Let-Off
        0xF801F1, // Fn37 One Shot Bell 1
        0xF801F2, // Fn38 One Shot Bell 2
        0xF801F3, // Fn39 One Shot Bell 3
        0xF801E1, // Fn40 Quilling Whistle Level 1
        0xF801E4, // Fn41 Quilling Whistle Level 2
        0xF801E8, // Fn42 Quilling Whistle Level 3
        0xF801EF, // Fn43 Quilling Whistle Level 4

For the existing functions F28-F31, the "delayed" start up and shut down commands can be kind of annoying. If we want to use the delayed start up function, the Lionel remote sends the delayed startup command (F28) and then the immediate startup right after (F29). A similar thing occurs for the delayed extended shutdown process: F30 first, and then F31. 

However, I noticed there is an issue with putting these legacy functions in the higher function range. Engine driver only supports up to 30 functions; is there a workaround for Lionel users? I personally think F22 and F23 could be reused for another function if we are limited by the GUI.

bobjacobsen, 2024-07-12:
Do you have the higher-numbered functions working in JMRI itself?  If so, and the only thing that's missing is EngineDriver support, the next test might be to open an issue on the [Engine Driver tracker](https://github.com/JMRI/EngineDriver/issues). I'm not sure how closely that's tracked, though.

tomixnscale89, 2024-07-12:
Yeah, the higher number functions are working in JMRI itself.

I'll create an issue on the Engine Driver repo.

What are your thoughts on changing how the extended startup/shutdown commands are sent?

bobjacobsen, 2024-07-12:
I think I can modify the code so you can specify two commands to send, e.g.:

```
       0xF801FBF801FC, // Fn28 Start Up Sequence 1 (Delayed Prime Mover, then Immediate Start up)
 ```

so that (in this case) Fn28 will send 0xF801FB immediately followed by F801FC.  That'll let you specify two commands for any of the function buttons you want.

Would that be sufficient?

bobjacobsen, 2024-07-12:
Should I be repeating both of those command messages while the button is down?  Or do they just get sent once?

tomixnscale89, 2024-07-12:
> I think I can modify the code so you can specify two commands to send, e.g.:
>
>
>
> ```
>
>        0xF801FBF801FC, // Fn28 Start Up Sequence 1 (Delayed Prime Mover, then Immediate Start up)
>
>  ```
>
>
>
> so [...]

I think that would be perfect. If you can set up the code this way, it would be a foundation for the Multi-Word Legacy commands. Those have 3 Words to send. 

You only need to send the each command once.

bobjacobsen, 2024-07-12:
By "three words", do you mean three commands, like 0xF81234, 0xF84567, 0xF87890?  The system I was thinking of doesn't really deal with that, as 0xF81234F84567F87890 is too big to fit in a Java long word (8 bytes).

Or do you mean something different?

tomixnscale89, 2024-07-12:
The third command (word) would be a checksum, so what you have (layering two commands into a Java long word) should work fine.

bobjacobsen, 2024-07-13:
The fix is in a branch in my GitHub repository at:

https://github.com/bobjacobsen/JMRI/tree/tmcc-double-functions

There are also installers here:

https://builds.jmri.org/jenkins/job/testreleases/job/bobjacobsen/130/

Let me know if it's doing what you want, please, and I'll have it merged to the main repository.

tomixnscale89, 2024-07-14:
Bob,

Sorry for the delay. Just gave it a try and it works great. Thank you again.

tomixnscale89, 2024-07-15:
I was working on implementing the multi-word commands with the checksum, but keep running into issues with the serial traffic controller. It seems that if the opcode is "FB", the traffic controller doesn't let all of the commands through.

In my code here, it ends up sending the third word which is the check sum, but not the rest. 

[image]

```java
if ( SERIAL_FUNCTION_CODES[func] > 0xFFFF ) {
                // TMCC 2 format
                if (SERIAL_FUNCTION_CODES[func] > 0xFFFFFF ) {
                    int first =  (int)(SERIAL_FUNCTION_CODES[func] >> 24);
                    int second = (int)(SERIAL_FUNCTION_CODES[func] & 0xFFFFFF);
                    // doubles are only sent once, not repeating
                    sendOneWordOnce(first  + address.getNumber() * 512);
                    sendOneWordOnce(second + address.getNumber() * 512);     
                    if((first & 0xFFFF) >= 0x170 && (first & 0xFFFF) <= 0x17F) // if the lower 9 bits are within 0x170 and 0x17F, this is a Multi-Word command.
                    {
                        sendOneWordOnce(createMultiWordChecksum(first,second));
                    }
                           
                } 
                else {
                    // single message
                    sendFnToLayout((int)SERIAL_FUNCTION_CODES[func] + address.getNumber() * 512, func);
                }
..........................................

// Example of Multi-Word Command
0xF8017CFB0010L, // Fn44 Front Pantograph Raised

................................

    // Generate the checksum for the previous two word's data parameters.
    private int createMultiWordChecksum(int word1, int word2) {
        
        word1 = word1  + address.getNumber() * 512;
        word2 = word2 + address.getNumber() * 512;
        
        // Multi word needs to calculate a checksum. 
        int byte1 = word1 & 0xFF;
        int byte2 = (word1 & 0xFF00) >> 8;
        int byte3 = word2 & 0xFF;
        int byte4 = (word2 & 0xFF00) >> 8;
        int byte5 = byte4; // a bit redundant
        
        int checksum = ~((byte1 + byte2 + byte3 + byte4 + byte5) % 256) ;
        return (0xFB0000 | (word2 & 0xFF00)) |  (checksum & 0xFF); // Make sure checksum is truncated by 0xFF
        
    }
```

bobjacobsen, 2024-07-15:
The constructor of a TMCC serial message from an int variable only takes the bottom 16 bits as the value:

```
    public SerialMessage(int value) {
        super(3);
        if ( (value & 0xFF0000) != 0) {
            setOpCode((value >> 16 ) & 0xFF);
        } else {
            setOpCode(0xFE);
        }
        putAsWord(value);
        setTimeout(100);
    }
```
where `putAsWord(..)` uses the bottom 16 bits as that word.  I think you need another constructor that can make a longer message.

tomixnscale89, 2024-08-04:
Bob,

Sorry for my extremely late reply on this, I got caught up with other things. 

Could you please elaborate more, I don't quite understand why my code wouldn't be working with putAsWord(...). I have it the same length as the other TMCC commands you implemented.

bobjacobsen, 2024-08-04:
I could be misunderstanding, but your code seems to convert `0xF8017CFB0010L` into `first = 0xF8017C` and `second = 0xFB0010`. Both of those are three bytes long, not two.

tomixnscale89, 2024-08-05:
Is my setup different than how you integrated the sequential commands from your branch? I tried to match my approach to yours:

`0xF801FBF801FCL, // Fn28 Start Up Sequence 1 (Delayed Prime Mover, then Immediate Start up)` 

from: [Line 102](https://github.com/bobjacobsen/JMRI/blob/tmcc-double-functions/java/src/jmri/jmrix/tmcc/SerialThrottle.java).

`0xF8017CFB0010L, // Fn44 Front Pantograph Raised`

The only difference apart from the creation of the checksum function is adding this snippet of code below after line 64 in your branch of [SerialThrottle.java](https://github.com/bobjacobsen/JMRI/blob/tmcc-double-functions/java/src/jmri/jmrix/tmcc/SerialThrottle.java):

```java
if((first & 0xFFFF) >= 0x170 && (first & 0xFFFF) <= 0x17F) // if the lower 9 bits are within 0x170 and 0x17F, this is a Multi-Word command.
{
  sendOneWordOnce(createMultiWordChecksum(first,second));
}
```

bobjacobsen, 2024-08-05:
Thank you!  I think I misunderstand what you're trying to do.

How many bytes should this code put on the wire to the TMCC system?

```
                   sendFnToLayout((int)SERIAL_FUNCTION_CODES[func] + address.getNumber() * 512, func);
 ```

Is that three or two?

tomixnscale89, 2024-08-05:
Including the Op Code, it should be three bytes.

For the Multi-Word command, it uses the same logic as sending two commands sequentially like you implemented a few weeks ago. The only change is a third command is sent to the track, which is a checksum of the first two commands.

danielb987, 2024-09-23:
@bobjacobsen 
This issue seems to have been forgotten for some time.

bobjacobsen, 2024-09-23:
@danielb987:

The changes @tomixnscale89 look valuable.  It would be good to get them fixed and included in JMRI.  Unfortunately, I don't really understand enough about TMCC to really help with this.

tomixnscale89, 2024-09-23:
Bob,

My code implementation should include what we need, but it seems to have some type of issue with the serial traffic controller, which I'm not familiar with. Perhaps I could push my fork of JMRI and you could build on your machine to see the error?

It's been a while since I did this, but from what I recall as long as you have some type of COM Port to send data to, the issue would come up. Don't need a Lionel system to test.

bobjacobsen, 2024-09-23:
Would be happy to do that!  Could you also give me a step-by-step for causing the problem, e.g. "open a throttle, select 27, hit function 12" or whatever will trigger the problem?  

Thanks

tomixnscale89, 2024-11-26:
Hey Daniel, 

My apologies, got quite busy past few months and didn't get around to this. I'll see about making some time at the end of the week and get a build up for Bob to try.

tomixnscale89, 2024-12-16:
Hi all,

Here is a branch that contains my local changes to SerialThrottle.java. 

[https://github.com/tomixnscale89/JMRI](url)

Try following these instructions to reproduce the error:

1. Launch JMRI from NetBeans IDE
2. When "Set Active Profile" screen popsup, create a new profile.
3. Select Profile then hit OK
4. Open the JMRI "Preferences" window. Change connection1 to "Lionel TMCC."
5. Change "System Connection" to "TMCC via Serial."
6. Select your Serial Port, then click "Save" on the bottom left.
7. Click "Restart" on Decoder Pro
8. Relaunch JMRI from NetBeans IDE
9. Select the Profile you created in step 2, then hit OK.
10. On the main DecoderPro window, click "Open Roster"
11. Click "New Loco."
12. For decoder, pick NMRA standard CV Definitions.
13. For address, set it to "3" with Short address and Roster ID. Then click save.
14. Select loco in the roster, then click "Throttle."
15. Click View -> "Reset Function Buttons." Functions up to 68 should now show up.
16. Locate FN44, FN45, FN46, or FN47. Try and activate them. 
17. The Output log in NetBeans for JMRI should give errors from the SerialTrafficController thread.

tomixnscale89, 2025-01-03:
Bob,

I hope you had a nice holidays and new year!

What I'm seeing is the following "return short message", along with the timeouts you saw:

[image]

I don't see the return short messages with any other commands, only the extended commands that I implemented.

tomixnscale89, 2025-01-07:
Last time I tried this, nothing seemed to work. However, I just tried it today, and everything does appear to be working. Really strange....

In that push from a few weeks ago, in addition to the TMCC2 3 Word support, I added another baud rate option (115200) for the new Lionel Base3's USB port in **SerialDriverAdapter.java.** 

When I select the new 115200 baud rate, the commands don't seem to be sent through to the base, I just get continuous timeouts in the terminal:
`20:47:35,044 mri.jmrix.AbstractMRTrafficController WARN  - Timeout on reply to message: FE 1F 9C consecutive timeouts = 0 in tmcc.SerialTrafficController [tmcc.SerialTrafficController Transmit thread]`

Perhaps I'm missing this addition somewhere else in the core part of JMRI's code? To recreate what I'm seeing, you can follow these steps:

    1. Launch JMRI from NetBeans IDE
    2. When "Set Active Profile" screen popsup, create a new profile.
    3. Select Profile then hit OK
    4. Open the JMRI "Preferences" window. Change connection1 to "Lionel TMCC."
    5. Change "System Connection" to "TMCC via Serial."
    6. Select your Serial Port, click "Additional Connection Settings", followed by changing the baud rate to 115200. Then click "Save" on the bottom left.
    7. Click "Restart" on Decoder Pro
    8. Relaunch JMRI from NetBeans IDE
    9. Select the Profile you created in step 2, then hit OK.
    10. On the main DecoderPro window, click "Open Roster"
    11. Click "New Loco."
    12. For decoder, pick NMRA standard CV Definitions.
    13. For address, set it to "3" with Short address and Roster ID. Then click save.
    14. Select loco in the roster, then click "Throttle."
    15. Click FN2 for the Horn.
    16. The Output log in NetBeans for JMRI should show consecutive timeout errors.

bobjacobsen, 2025-01-07:
Your 115200 changes seem to work.  The port is opening and transmitting at that speed.

The timeouts probably mean that the Command Station is not replying.  Is there some option internal to the Command Station that has to be set to 115200?

tomixnscale89, 2025-01-08:
Not from my experience in writing other software to interact with the Base. Everything should work with:
115200 Baud
No Parity
8 bits
No Handshake

bobjacobsen, 2025-01-17:
Sorry for the slow reply, busy time.

I can't think of anything more useful that I can do to debug the 115200 problem, unfortunately.  I've got the most recent content of your branch, and it seems to be setting the port up fine for 115200.  

Does your other software work through the same serial adapter?  Maybe that's the problem.

PR #13258 (2024-07-15): Add double-function sending to TMCC support
See #13253 for background.  Closes #13253.