JMRI/JMRI #13262: JMRI cannot read Digitrax SDN144A1 speed table

BigjoeH, 2024-07-18:
I installed and set up JMRI Decoder Pro over the past weekend.  It's on my Windows 10 desktop and connected to my program track via a Digitrax PR4 interface.  It seems to work well with on exception:  it cannot read the speed table on my Digitrax SDN144A1 decoder.  Digitrax says I'm on my own because they don't support 3rd party software.  So I'm asking you:  Any ideas about how to fix this problem?  Thanks.

bobjacobsen, 2024-07-18:
Could you be a little more specific about what "cannot read the speed table" means?  Do you get error messages?  Or does the speed table not appear?  Or something else?

BigjoeH, 2024-07-18:
The table does not appear.  A message appears saying something generic about what I'm looking for might be elsewhere.  The same message appears for the motor tab and one or two others.  Yet I can read a TCS decoder clearly including the speed table.  Thanks for asking.

H

bobjacobsen, 2024-07-18:
This sounds like you've ended up with the wrong decoder definition somehow.

Could you try the following please?
 - Create a new roster entry with the New Loco button
 - When it shows the window to select a decoder, select
   - Digitrax
   - Then SFX Sound Decoders (3rd choice, I think)
   - The the SDN144A1 decoder
 - Then click Open Comprehensive Decoder

I think that will make sure you've got the right decoder definition.

BigjoeH, 2024-07-18:
Thanks.  I'll try it.  But is there no way to simply change the decoder ID in the already established roster listing?

H

bobjacobsen, 2024-07-18:
>  But is there no way to simply change the decoder ID in the already established roster listing?

No, there’s no way to do that.  In general, there’s no way to convert one decoder definition to another. They’re just too different.

BigjoeH, 2024-07-18:
Thanks, again.  And, by the way, your suggestion to create a new roster entry worked.  I had to delete the old entry first, but now every page of Decoder Pro appears as expected.  🤠

H

closed 2024-07-20 by BigjoeH