JMRI/JMRI #13281: Use of system name/user name by LogixNG in local variables

jerryg2003, 2024-07-27:
I have the following LogixNG to publish an MQTT message when an RFID Report comes in to JMRI:

LogixNG: MQTT: Train ID
    ConditionalNG: Reporter G ::: Startup
        ! A
            If Then Else. Execute on change
               ::: Local variable "TrainID", init to None "[init]"
                ? If
                    Reporter MRE32-RFID08/G Current Report is not null
                ! Then
                    Many
                        ! A1
                            Set variable TrainID to the current report of reporter MRE32-RFID08/G (No Listen)
                        ! A2
                            Publish by local variable "TrainID" to topic jSend/rpt/G. Retain: Default
                        ! A3
                            Log local variables 

It works the way I want; for example:  if JMRI sees the MQTT reporter message with RFID tag "045B89D222", an MQTT message will be published with the message value of "CN3745" which is the user name assigned to that RFID tag.

However, the "log local variables" shows the following:

15:40:17,246 rit.logixng.actions.LogLocalVariables WARN  - Log local variables: [JMRI LogixNGThread]
15:40:17,247 rit.logixng.actions.LogLocalVariables WARN  -     Name: TrainID, Value: ID045B89D222 [JMRI LogixNGThread] 

Tried this with TrainID local variable type of "none" and "string" - same result.

Shouldn't the log of local variable TrainID show "CN3745" since that is what it must be since it shows up in the published MQTT message?

BTW, what is the impact of Listen/No Listen on the statement that sets the local variable from the Reporter?  Doesn't seem to change the behavior when I try.  [As opposed to changing to "No Listen" on the If statement  - this doesn't LogixNG doesn't work so that is clearly necessary.]

danielb987, 2024-07-29:
@dsand47 
The action `Log local variables` calls the method [SymbolTable.printVariable()](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/logixng/SymbolTable.java#L255) to print each local variable. The method `printVariable()` calls `log.warn()` to print the value. And `log.warn()` calls `value.toString()` to get the value as a string.

The method [AbstractNamedBean.toString()](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/implementation/AbstractNamedBean.java#L257) always returns the system name and optional an additional string. This is due to that Swing classes depends on that the value of `toString()` doesn't change, or at least that the beginning of the value doesn't change.

The method `SymbolTable.printVariable()` has a special handling for maps and lists. Should it also have a special handling for NamedBeans so that it prints the `getDisplayName()` instead of the system name?

bobjacobsen, 2024-07-29:
> And log.warn() calls value.toString() to get the value as a string.

You can call value.getDisplayName() and pass the result into the log.warn(..) statement.

dsand47, 2024-07-29:
Jerry,

Since you are working with a RFID report, the contents of "Current Report" is an ID Tag object ( such as ID045B89D222), not a string value.  The ID Tag object has the value, in this case CN3745.

When the publish command is run, it does a "toString" for ID045B89D222 which returns CN3745.

Dave Sand

jerryg2003, 2024-07-30:
Thanks all.  I understand what is happening now.  

As a “naive” user, I would expect the “log local variable” to show the same as is actually in the local variable.  In this case, if TrainID is specified to be a string, then the display name (again, in this case the user name) should be in the log since that is what is in the local variable.  If TrainID is specified as type “none,” then I guess object name would be technically correct, although it still shows up as the display name when used in the publish statement, so why should the log show something different?  I would vote for display name in this case too.

 Again, just my 2 cents from POV of someone trying to use the log to help me debug my LogixNGs.

dsand47, 2024-07-30:
A local variable changes its "class" depending on the content.  While TrainID may have been defined as a string, when an ID Tag object is assigned to the variable, the class is changed.  It is no longer a string type.

There is one exception.  If the LogixNG preference is set to "Use strict typing", the class will not change.  In that case an exception will occur when the content's "class" does not match the variable's "class".

----

I have confirmed that an extra space is added when the ID Tag is used, but not for direct reporter publishing via the LogixNG.

I have figured out how to do a round trip using a command line publish to a JMRI reporter and a command line subscription to get the result back and print the hex values of the returned message.  The result of a 24 hour crash course in MQTT, JMRI Reporters and ID Tags.

Since I don't have any hardware, MQTT will work for investigating the RFID issues since it also creates ID Tags.

dsand47, 2024-07-31:
The extra space in the rpt content is caused an incomplete payload to track/reporter.  According to the MQTT Help, the format is "nnnn text".

jerryg2003, 2024-07-31:
JMRI does not publish any messages when it sees a change in a reporter so this only applies to messages subscribed by JMRI.  [BTW, I wrote the MQTT Help and it should say "abcd[ optional other text]" as an acceptable input for a reporter message. Thanks for pointing this out. I'll fix.]  

Here is an example of the incorrect character count being generated by LogixNG.  Note that other JMRI-generated messages, for example, when sensor or light changes, have the correct character count.   [Note that I have set up different topics in JMRI Preferences for send and receive (helps to keep track of what is going on, especially when debugging the layout).  I strongly recommend that, at least for developers.]

The messages with "jSend" are all generated by JMRI.  The ones generated via a LogixNG ("disp" in the topic) have the added space and character count. However, those generated by JMRI on a change to a light ("lt") and a sensor ("sens") have the correct character count.  

2024-07-30T15:58:06-0400        jxtrains/jSend/disp/E32-RFID08/G 0 0 7                  CN4375
2024-07-30T15:58:08-0400        jxtrains/jSend/disp/E32-RFID08/G 0 0 7                  CN4375
2024-07-30T15:58:10-0400          jxtrains/jSend/lt/E32-RFID08/0 0 0 3                      OFF
2024-07-30T15:58:10-0400      jxtrains/jSend/sens/E32-RFID08/ACK 0 0 8                 INACTIVE

dsand47, 2024-07-31:
The character count issue is caused by the MQTT Recorder code.  It is not caused by LogixNG.  If you add a reporter to a block and create a one word MQTT payload, the block value will have a space at the end.

The MQTT Reporter payload is split into two pieces.  The first word is the "loco".  It is used to create the ID Tag system name which will be "<prefix><first word>".  Prefix depends on the connection and type, such as MR for MQTT Reporter.  The rest of the payload will be the "content".  This will contain zero or more space delimited words.  The "content" is attached to the ID Tag as a "content" property.

When a request is made for a reporter's current or last report and the field contains an IDTag instead of a string, "toReportString" is called.  

"toReportString" starts by using the user name if available.  If not it uses the system name minus the prefix.  It then appends a space and the content for each property key.  There is always a "content" property for MQTT Reporters, which might have no data.  When there is no data, the value returned has a space at the end.

I don't know if other sources of IDTags have the same issue.  I don't have RFID, RailCom or Digitrax Transponder hardware.

dsand47, 2024-07-31:
@danielb987 
> The method `SymbolTable.printVariable()` has a special handling for maps and lists. Should it also have a special handling for NamedBeans so that it prints the `getDisplayName()` instead of the system name?

It should be ok since it is limited to displaying content.

Something like the following?
```
diff --git a/java/src/jmri/jmrit/logixng/SymbolTable.java b/java/src/jmri/jmrit/logixng/SymbolTable.java
index d6685e7074..6cbadfbb25 100644
--- a/java/src/jmri/jmrit/logixng/SymbolTable.java
+++ b/java/src/jmri/jmrit/logixng/SymbolTable.java
@@ -285,6 +285,9 @@ public interface SymbolTable {
             String className = showClassName && value != null
                     ? ", " + value.getClass().getName()
                     : "";
+            if (value instanceof jmri.implementation.AbstractNamedBean) {
+                value = ((jmri.NamedBean)value).getDisplayName();
+            }
             log.warn("{}{}: {}, {}: {}{}", pad, headerName, name, headerValue, value, className);
         }
     }
```

danielb987, 2024-08-03:
> It should be ok since it is limited to displaying content.
> 
> Something like the following?

Thank you! See #13297.

closed 2024-08-04 by danielb987