JMRI/JMRI #13292: Dispatch

fert75, 2024-08-01:
Hi. I made the same layout with virtual head signals (I don't have a signals hardware). When I try to do a Dispatch appears the following errors: "Found 6 error when checking/setting direction Sensors in signal logic for this Transit. Check Console log for details." In the Console log: "Warn - Section - West main 1 (IY:Auto:0003) , 2 turnouts missing signals ([ANT-EventQue-0]". Four rows errors One for each Section. Please help me. Thanks.

[image]
[image]

dsand47, 2024-08-01:
Since you are using signal heads, you need to add signal heads to the turnouts and anchor points that are block boundaries.

If you were to use signal masts, the requirements are more flexible.

Dave Sand

IMG_3273.jpg (view on web) <https://github.com/user-attachments/assets/4dcfa814-7bcf-44cd-a485-b88e5457a98d>
IMG_3274.jpg (view on web) <https://github.com/user-attachments/assets/3660047a-4061-411e-abd3-ed8786f6b0b2>

dsand47, 2024-08-02:
The xml file was not attached or there was a delivery issue.

Upload the xml file to the group's ProblemsBeingWorkedOn file folder.

Dave Sand

fert75, 2024-08-02:
Enviado desde Correo para Windows De: Dave SandEnviado: viernes, 2 de agosto de 2024 13:26Para: JMRI/JMRICC: fert75; AuthorAsunto: Re: [JMRI/JMRI] Dispatch (Issue #13292) The xml file was not attached or there was a delivery issue. Upload the xml file to the group's ProblemsBeingWorkedOn file folder. Dave Sand ----- Original message ----- From: fert75 ***@***.***> To: JMRI/JMRI ***@***.***> Cc: Dave Sand ***@***.***>, Comment ***@***.***> Subject: Re: [JMRI/JMRI] Dispatch (Issue #13292) Date: Friday, August 02, 2024 8:00 AM Hi Dave! I send my layout xlm file. Could you help me Please to fix the the console Warm issues? Thanks! Fernando Enviado desde Correo para Windows De: Dave SandEnviado: jueves, 1 de agosto de 2024 18:58Para: JMRI/JMRICC: fert75; AuthorAsunto: Re: [JMRI/JMRI] Dispatch (Issue #13292) Since you are using signal heads, you need to add signal heads to the turnouts and anchor points that are block boundaries. If you were to use signal masts, the requirements are more flexible. Dave Sand ----- Original message ----- From: fert75 ***@***.***> To: JMRI/JMRI ***@***.***> Cc: Subscribed ***@***.***> Subject: [JMRI/JMRI] Dispatch (Issue #13292) Date: Thursday, August 01, 2024 1:42 PM Hi. I made the same layout with virtual head signals (I don't have a signals hardware). When I try to do a Dispatch appears the following errors: "Found 6 error when checking/setting direction Sensors in signal logic for this Transit. Check Console log for details." In the Console log: "Warn - Section - West main 1 (IY:Auto:0003) , 2 turnouts missing signals ([ANT-EventQue-0]". Four rows errors One for each Section. Please help me. Thanks. IMG_3273.jpg (view on web) <https://github.com/user-attachments/assets/4dcfa814-7bcf-44cd-a485-b88e5457a98d> IMG_3274.jpg (view on web) <https://github.com/user-attachments/assets/3660047a-4061-411e-abd3-ed8786f6b0b2>

dsand47, 2024-08-02:
Sorry, I thought the message was in the user group.

To attach a file in GitHub you need to change the `.xml` suffix to `.txt`.

fert75, 2024-08-02:
<?xml version="1.0" encoding="UTF-8"?>
<?xml-stylesheet href="/xml/XSLT/panelfile-5-5-5.xsl" type="text/xsl"?>
<layout-config xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/layout-5-5-5.xsd">
  <jmriversion>
    <major>5</major>
    <minor>8</minor>
    <test>0</test>
    <modifier />
  </jmriversion>
  <sensors class="jmri.jmrix.dccpp.configurexml.DCCppSensorManagerXml">
    <globalDebounceTimers>
      <goingActive>0</goingActive>
      <goingInActive>2000</goingInActive>
    </globalDebounceTimers>
    <sensor inverted="false">
      <systemName>D4S22</systemName>
      <properties>
        <property>
          <key>Type</key>
          <value class="java.lang.String">SENSOR</value>
        </property>
        <property>
          <key>Pin</key>
          <value class="java.lang.Integer">22</value>
        </property>
        <property>
          <key>Pullup</key>
          <value class="java.lang.Boolean">true</value>
        </property>
        <property>
          <key>ID</key>
          <value class="java.lang.Integer">22</value>
        </property>
      </properties>
      <debounceTimers>
        <goingActive>0</goingActive>
        <goingInActive>3000</goingInActive>
      </debounceTimers>
    </sensor>
    <sensor inverted="false">
      <systemName>D4S23</systemName>
      <properties>
        <property>
          <key>Type</key>
          <value class="java.lang.String">SENSOR</value>
        </property>
        <property>
          <key>Pin</key>
          <value class="java.lang.Integer">23</value>
        </property>
        <property>
          <key>Pullup</key>
          <value class="java.lang.Boolean">true</value>
        </property>
        <property>
          <key>ID</key>
          <value class="java.lang.Integer">23</value>
        </property>
      </properties>
      <debounceTimers>
        <goingActive>0</goingActive>
        <goingInActive>3000</goingInActive>
      </debounceTimers>
    </sensor>
    <sensor inverted="false">
      <systemName>D4S24</systemName>
      <properties>
        <property>
          <key>Type</key>
          <value class="java.lang.String">SENSOR</value>
[... 1300 lines cut ...]
      </inactive>
      <unknown url="program:resources/icons/smallschematics/tracksegments/circuit-error.gif" degrees="0" scale="1.0">
        <rotation>0</rotation>
      </unknown>
      <inconsistent url="program:resources/icons/smallschematics/tracksegments/circuit-error.gif" degrees="0" scale="1.0">
        <rotation>0</rotation>
      </inconsistent>
      <iconmaps />
    </sensoricon>
    <sensoricon sensor="D4S29" x="360" y="510" level="10" forcecontroloff="false" hidden="no" positionable="true" showtooltip="true" editable="true" momentary="false" icon="yes" class="jmri.jmrit.display.configurexml.SensorIconXml">
      <active url="program:resources/icons/smallschematics/tracksegments/circuit-occupied.gif" degrees="0" scale="1.0">
        <rotation>0</rotation>
      </active>
      <inactive url="program:resources/icons/smallschematics/tracksegments/circuit-empty.gif" degrees="0" scale="1.0">
        <rotation>0</rotation>
      </inactive>
      <unknown url="program:resources/icons/smallschematics/tracksegments/circuit-error.gif" degrees="0" scale="1.0">
        <rotation>0</rotation>
      </unknown>
      <inconsistent url="program:resources/icons/smallschematics/tracksegments/circuit-error.gif" degrees="0" scale="1.0">
        <rotation>0</rotation>
      </inconsistent>
      <iconmaps />
    </sensoricon>
    <sensoricon sensor="D4S30" x="550" y="510" level="10" forcecontroloff="false" hidden="no" positiona

dsand47, 2024-08-02:
You cannot attach a file using email.  You need to do it from the GitHub web site.

closed 2024-10-28 by github-actions[bot]