JMRI/JMRI #13294: Dispatcher issue

fert75, 2024-08-02:
[TEST2_FT.txt](https://github.com/user-attachments/files/16474127/TEST2_FT.txt)

Hi! I made this layout with virtual head signals (I don't have a signals hardware). When I try to do a Dispatch appears the following errors: "Found 6 error in the System Console:
"Section East main 1 (IY:AUTO:0004), 1 anchor point signal heads missing [ANT-EventQue-0]
"Section North main 1 (IY:AUTO:0002), 3 anchor point signal heads missing [ANT-EventQue-0]
"Section West main 1 (IY:AUTO:0003), 2 anchor point signal heads missing [ANT-EventQue-0]

I send a XML file with all tables content and panels.
I don't know how I can solve it. I'm a beginner in this software.

Thanks!

fert75, 2024-08-02:
> I looked at your xml file and there is a serious issue.
> The issue is that most of the track segments are joined using edge connectors instead of anchor points.  Edge connectors are used to connect tracks on separate layout editor panels.
> I recommend that you delete the panel and create a new one using anchor points.
> For the crossovers, I recommend using the Layout Editor crossover items instead of separate turnouts.
> Is there a reason for using signal heads?  You already have a block occupancy topology that is optimized for signal masts using signal mast logic.  This will also make it easier to get Dispatcher working.

Thanks for your help! For the crossover what kind of signal masts do you recommend? Could you bring me an example to how should be?Thanks?

fert75, 2024-08-03:
Hi, Dave, I have followed all your suggestion and I solved same issues. but only I need to fix these:   I have upload the new XML file in the group folder. Thanks Fernando Enviado desde Correo para Windows De: Dave SandEnviado: viernes, 2 de agosto de 2024 18:06Para: JMRI/JMRICC: fert75; AuthorAsunto: Re: [JMRI/JMRI] Dispatcher issue (Issue #13294) I looked at your xml file and there is a serious issue.The issue is that most of the track segments are joined using edge connectors instead of anchor points. Edge connectors are used to connect tracks on separate layout editor panels.I recommend that you delete the panel and create a new one using anchor points.For the crossovers, I recommend using the Layout Editor crossover items instead of separate turnouts.Is there a reason for using signal heads? You already have a block occupancy topology that is optimized for signal masts using signal mast logic. This will also make it easier to get Dispatcher working.

dsand47, 2024-08-03:
The track connections are now correct.  The signal masts and signal mast logic sort of work.  I recommend that you do not disable aspects for now.  Doing so can result in unexpected results, such as displaying a Restricted aspect instead of the expected Clear aspect.  The signal system and mast types affect the signal mast logic.

You have your transits starting in sections that are turnout blocks.  You need to start them in regular block sections, such as "South main 1 B24" and "South main 2 B30".  If you plan to have the trains run continuously, the first section is repeated as the last section.

fert75, 2024-08-04:
Excellent!  What about the issues in System console that it not allow to run Dispatcher?Enviado desde mi iPhone

dsand47, 2024-08-04:
> Excellent!  What about the issues in System console that it not allow to run Dispatcher?

Are you still getting errors?  The original errors were related to the required direction sensors when using SSL.   Signal masts with SML do not need direction sensors.

You do need to change the Dispatcher options with the move to signal masts.

fert75, 2024-08-05:
Hi Dave, finally I have been able to solve the issues. Thanks for your help!Another question is about block occupancy. Can I use a infrared sensor instead of current detection?I upload a xml file for your suggestion to improve my layout. Thanks so much!

dsand47, 2024-08-05:
> Another question is about block occupancy. Can I use a infrared sensor instead of current detection?

The issue with infrared sensors is that they respond to a train in a specific spot.  They can turn on and off as a train is moving.  

Dispatcher absolutely depends on consistent occupancy detection.

fert75, 2024-08-05:
Dave,

In the transit panel>>actions I have found an option to set sensor active
or inactive when entry to the section and exit from this section. Could I
solve it when the train is in a specific spot?

dsand47, 2024-08-05:
> Dave, In the transit panel>>actions I have found an option to set sensor active or inactive when entry to the section and exit from this section. Could I solve it when the train is in a specific spot?

No.  The entry or exit of a section is triggered by block occupancy.  

Some people have tried to use a set of infrared sensors but there are always gaps which result in inconsistent occupancy.

fert75, 2024-08-06:
Dave, what kind of occupancy sensor do you recommend? What about ACS712 for arduino?

dsand47, 2024-08-06:
> Dave, what kind of occupancy sensor do you recommend? What about ACS712 for arduino?

You need to ask that question on the JMRI user group.  I use LCC/OpenLCB.

fert75, 2024-08-11:
Hi, how can I use a sensor to activate a servo level crossing?

dsand47, 2024-08-11:
> Hi, how can I use a sensor to activate a servo level crossing?

This is another user group question.  

In JMRI a "level crossing" is a diamond.  There are no turnouts involved.  It sounds like you might be talking about a "crossover" between parallel tracks.

closed 2024-11-11 by github-actions[bot]