JMRI/JMRI #13316: DecoderPro CV read timeouts for NCE PowerPro PH5

markelliot123, 2024-08-09:
It appears that the JMRI DecoderPro times out too quickly for engine decoders using the NCE PowerPro PH5 system with the built in USB. This is especially an issue with older engine decoders. Sometimes it can work and sometimes not. Mostly not when it fails. You can tell the engines are receiving the commands because they growl and creep along the program track when the CV reads are sent. At the same time, the Power Cab can flawlessly reads and write the CV registers on the same engine and program track. Yes, I've tried JMRI with the Power Cab throttle disconnected too. Yes, I have all of the latest Windows 11 updates and USB drivers. I have JMRI version 5.8 and Java 11.  What is interesting is the PanelPro works great. I can throw switches, set routes and run engines from JMRI throttles. However, the same preference settings do not work in DecoderPro. Please advise if there are any known work-arounds or solutions.

markelliot123, 2024-08-09:
One thought I had. Is there anyway to set the read time out value? It would be very handy to optionally set this either globally or by engine roster in the PanelPro. That maybe a simple way out for everyone. Assuming that is the issue.

markelliot123, 2024-08-22:
I finally figured this out. The default "Mode" for DecoderPro is "Direct". This does work on my newest engines but not consistently for my older engines. True JRMI does document this if you search long and hard for it. **However, there IS a JMRI bug!!! The bug is there is no way to save this mode setting in the engine roster.** So, you have to remember to do this each time and for specific engines in the roster. A frustrating problem especially if you only do decoder changes every few months. What is strange is I don't remember having to worry about this in past versions of JMRI. I've attached a screen shot of what I'm talking about.
[image]

KenC57, 2024-10-07:
Two questions come to mind here. One is if it was ever designed that a roster entry retained the programming mode. I don't think that was ever a design feature. The second is if the system should try different mode when first talking with a decoder. I thought in the identify feature it would start with direct but then also try paged if direct failed.

markelliot123, 2024-10-15:
Anything automatic would be great. My issue is I program an engine once
every few years per engine. Settings stay the same for me for a long time.
It's easy to forget what to do, if something does not work.

PUlvestad, 2024-12-09:
The roster entry gets it programming mode from the decoder definition. It is not stored in the roster entry.

    <programming direct="yes" paged="yes" register="yes" ops="yes"/>

One solution is to modify 'your' definition by changing yes in direct to no and saving the modified definition to your preference folder and then recreating the decoder index. This prevents the modified file from reverting when JMRI is updated.

[image]

[image]