JMRI/JMRI #13325: Turnouts window does not collapse properly

rm5248, 2024-08-13:
At the bottom of the 'Turnouts' window, there are some checkboxes on the bottom:
[image]

These do not collapse based on the size of the window, so they can be rendered off-screen if the window is too small.

danielb987, 2024-08-15:
@icklesteve 
The panel with the checkboxes should probably have a `jmri.util.swing.WrapLayout` to fix it.

rm5248, 2024-08-15:
Suggestion: a drop down to let you select the different options might make sense to keep the layout similar.  I don't think there's anything like that readily available in Java though.

Pugwash1, 2024-08-15:
It would be nice to get rud of the check box's, use right click on column header to show or display (it might already be there) and a blank line for adding, like any normal 21st century grid works.

icklesteve, 2024-08-24:
Hi folks, see #13374
Not 100% sure where a dropdown ( menu ) would fit here, there are already 2 methods of selecting which columns are visible, adding a 3rd may not help, column visibility can be selected via right click on the column header.
In a future PR it would be good to seperate column visibility from the Bean seetings, eg. moving Turnout Automatic Retry to the Retry Automation settings menu, however out of scope for #13374

closed 2024-08-25 by danielb987

icklesteve, 2024-08-26:
@rm5248 if you could check when your next update that the tables perform OK that would be great, thanks !

danielb987, 2024-08-26:
> @rm5248 if you could check when your next update that the tables perform OK that would be great, thanks !

If you want installers with this fix, you can find it here:
https://builds.jmri.org/jenkins/job/development/job/packages/

rm5248, 2024-08-27:
Looks good to me, thanks!

PR #13374 (2024-08-25): Beantables bottom bar
AbstractTableAction -
getPanel() now returns a JComponent ( so that a JTabbedPane does not have to be nested within a JPanel )

AbstractTableTabAction -
remove dataPanel , return JTabbedPane in getPanel()

BeanTables -
bottomBox to WrapLayout

fixes #13325