JMRI/JMRI #13331: A new WebPage for Webserver and Operations with RFID

nigelcliffe, 2024-08-15:
A Webpage for inclusion in the JMRI distribution,  place in 
...../JMRI/web/ 

[operations_RFID_Trains.zip](https://github.com/user-attachments/files/16623347/operations_RFID_Trains.zip)

The page presents an alternative for OperationsPro "Conductor" use.  The Javascript may be more accessible for subsequent modification than the existing Operations web pages.   The intention is to work alongside ID-Tags on Cars (wagons in UK), with the order of the Cars in the train presented based on their most recent ID-Tag reports.   

Many thanks to Steve Todd for advice on how to get this page working, and some changes to the underlying JMRI capabilities.

danielb987, 2024-08-16:
@mstevetodd 
Is this something for you to look at?

danielb987, 2024-08-16:
@nigelcliffe 
Thank you!

mstevetodd, 2024-08-16:
> @mstevetodd 
> Is this something for you to look at?

Certainly

mstevetodd, 2024-08-16:
@danielb987 or @bobjacobsen or anyone else,
Is there an appropriate place to put a link and a short description of this example?
Something similar to https://www.jmri.org/help/en/html/tools/scripting/Examples.shtml

Any suggestions appreciated.

nigelcliffe, 2024-08-16:
Updated code, after bug found (lack of table sorting for local moves). 

Also reduced number of "no sort" columns in the tables, so almost all columns are now sortable.  

[operations_RFID_Trains.zip](https://github.com/user-attachments/files/16641432/operations_RFID_Trains.zip)

bobjacobsen, 2024-08-16:
> Is there an appropriate place to put a link and a short description of this example?

@DanielBoudreau, do you have an opinion on this? It's an alternative to an operations page.

DanielBoudreau, 2024-08-16:
No opinion, but always happy to see enhancements.

Dan

danielb987, 2024-08-17:
@mstevetodd 
> @danielb987 or @bobjacobsen or anyone else, Is there an appropriate place to put a link and a short description of this example? Something similar to https://www.jmri.org/help/en/html/tools/scripting/Examples.shtml
> 
> Any suggestions appreciated.

See PR #13343

I suggest that you either merge my branch into your branch or that you simply copies my files in your branch. I'm about to start building the next JMRI test release today or tomorrow and it would be nice to have this included.

My PR adds the link `Web server tools...` to the left sidebar on the main JMRI web page, below the link `System-specific...` in the `Tools` section.

danielb987, 2024-08-17:
Also note that the `WebServerTools.shtml` in my PR is only some kind of template. Fill it with whatever you want.

nigelcliffe, 2024-08-17:
Suggested replacement for Line23 (description) inside  " help/en/html/tools/WebServerTools.shtml  "

<p>Alternate "Conductor" webpage, with Id-Tag updates (eg. RFID), and showing current train composition.   Illustrates use of the jQuery mechanisms into the JMRI webserver, and the displaying of information in dynamically updating and sortable tables.   To see "where last seen" and "when last seen" data from Id-Tag reports requires the following steps :  a) associate each Id-Tag with a Car (Wagon),  b) define Reporters (in the Reporter table) which receive the Id-Tag reports, c) associate each Reporter with a Location (either a Location in general, or a specific road/siding within the Location).    </p>

danielb987, 2024-08-17:
> Suggested replacement for Line23 (description) inside " help/en/html/tools/WebServerTools.shtml "

Thank you! Fixed.

PR #13336 (2024-08-18): web: alternate "Conductor" webpage, with Id-Tag updates (eg. RFID), and
showing current train composition.
This closes #13331 and #13343
closed 2024-08-18 by danielb987