JMRI/JMRI #13333: OpenLCB Firmware Upgrade window issues

rm5248, 2024-08-16:
There are two things that I've noticed with the firmware upgrade page:

1. You can press the "Load" button if you have not selected a target node ID.  This results in the following stack trace:
```
19:46:17,947 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException: null
	at org.openlcb.swing.NodeSelector.getSelectedItem(NodeSelector.java:268) ~[openlcb.jar:?]
	at jmri.jmrix.openlcb.swing.downloader.LoaderPane.destNodeID(LoaderPane.java:226) ~[jmri.jar:?]
	at jmri.jmrix.openlcb.swing.downloader.LoaderPane.doLoad(LoaderPane.java:175) ~[jmri.jar:?]
	at jmri.jmrix.AbstractLoaderPane.lambda$initComponents$1(AbstractLoaderPane.java:155) ~[jmri.jar:?]
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) ~[?:?]
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) ~[?:?]
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) ~[?:?]
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) ~[?:?]
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) ~[?:?]
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297) ~[?:?]
	at java.awt.Component.processMouseEvent(Component.java:6635) ~[?:?]
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3342) ~[?:?]
	at java.awt.Component.processEvent(Component.java:6400) ~[?:?]
	at java.awt.Container.processEvent(Container.java:2263) ~[?:?]
	at java.awt.Component.dispatchEventImpl(Component.java:5011) ~[?:?]
	at java.awt.Container.dispatchEventImpl(Container.java:2321) ~[?:?]
	at java.awt.Component.dispatchEvent(Component.java:4843) ~[?:?]
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) ~[?:?]
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547) ~[?:?]
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) ~[?:?]
	at java.awt.Container.dispatchEventImpl(Container.java:2307) ~[?:?]
	at java.awt.Window.dispatchEventImpl(Window.java:2772) ~[?:?]
	at java.awt.Component.dispatchEvent(Component.java:4843) ~[?:?]
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) ~[?:?]
	at java.awt.EventQueue$4.run(EventQueue.java:721) ~[?:?]
	at java.awt.EventQueue$4.run(EventQueue.java:715) ~[?:?]
	at java.security.AccessController.doPrivileged(Native Method) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) ~[?:?]
	at java.awt.EventQueue$5.run(EventQueue.java:745) ~[?:?]
	at java.awt.EventQueue$5.run(EventQueue.java:743) ~[?:?]
	at java.security.AccessController.doPrivileged(Native Method) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) ~[?:?]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:742) ~[?:?]
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) [?:?]
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) [?:?]
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) [?:?]
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) [?:?]
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) [?:?]
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:90) [?:?]
```

2. The "Target Node ID" box is  too small, I can only see the Node ID when attempting to select something.  The name of the node is very useful to see.
[image]

bobjacobsen, 2024-08-17:
What Look&Feel are you using?  E.g. Windows, Motif, etc?

On my Mac, when I click on the "Target Node ID:" field to select it, I get the full-width display:

[image]

Does that not work for you?  If not, I might have to write some GUI L&F-specific code for this.

(Have a fix for the exception, thanks for pointing that out)

rm5248, 2024-08-17:
Default(Metal) look and feel on Linux(opendjdk11).

If I open the window a second time it will go to a full width(actually large enough that it renders outside of the window).  I had to expand the window in order to take this screenshot.

[image]

Changing to the GTK L&F results in the same behavior.

The width of the data in the drop down is the same as the width of the drop down.

[image]

bobjacobsen, 2024-08-17:
Thanks! There's a fix for both of these in Openlcb/OpenLCB_Java#268. Once that's merged and a new version of the library made, we can include it in JMRI.  Might take a couple days.

bobjacobsen, 2024-08-21:
The fix has been merged.  It'll be in development builds starting tomorrow, and the next test release in about a month.

closed 2024-08-21 by bobjacobsen