JMRI/JMRI #13341: Lables and Meda

djcalcutt, 2024-08-17:
Issue Type: __Bug__

Unchecking Function so thaey are not visible in the throttle window, that have no lable, still show in the throlle window.

I placed a word 'Blank' in the lable and then unchecked the visible checkbox, then the Throttle window correctly removed the function from 

JMRI 5.0.2 on Jave 17 with windows 10

PanelPro 5.9.2+R37ac0ba483
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 17.0.11+7-LTS-207 mixed mode, sharing
Windows 10 10.0 amd64

Lionel642, 2024-08-18:
Yes, this is as coded, this visibility setting only applies when a label is defined. I'll update the documentation about that.

djcalcutt, 2024-08-18:
In my world…. which is no perfect. The empty  label set to visible, will show a function button as empty. Either way , im look for a way to remove function button from the web throttle. Too many general public people just try all functions not knowing what they do. Darrin J. CalcuttSent from my iPhone

djcalcutt, 2024-08-18:
I tried editing XML file. But this did not work. Aug 18, 2024 jrmi 4 | Synology Photosnas.cloudbusinessportal.caAny suggestion ?Darrin J. CalcuttSent from my iPhone

closed 2026-10-06 by Lionel642