JMRI/JMRI #13342: webhottle disregards Visibiliy Check

djcalcutt, 2024-08-17:
Issue Type: __Bug__

After limited the function people can see 

I uncheck the Visibiliy check box, and save,

Open the webthrottle for the locomotive and the function still exsist.

reopen the label and media window and confim the setting are still unchecked, so the lable should no me displaying

JMRI 5.9.2, Java 17, Windows 10

Here is me web thorttle http://atc.lovethosetrains.com:12080/web/webThrottle.html?loconame=1

functions are still showing

PanelPro 5.7.9+R80852804ee
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 11.0.20+9-LTS-256 mixed mode
Windows 10 10.0 amd64

djcalcutt, 2024-08-17:
PS WiThrolle still shows the function too... so is JMRI causing the WiThrolle to also show the unwanted functions?

mstevetodd, 2024-08-17:
@djcalcutt please provide a screenshot of the window showing where you are setting this "visibility check box".

djcalcutt, 2024-08-17:
Darrin J. CalcuttSent from my iPhone

djcalcutt, 2024-08-17:
One this i did notice when saving… since my roster is 1, should n’t is be 1.xml?I better check my roster. Darrin J. CalcuttSent from my iPhone

mstevetodd, 2024-08-17:
@djcalcutt 
I'm not seeing any screenshots. You probably need to attach them using the web interface, not by attaching them to a reply email.
--SteveT

djcalcutt, 2024-08-17:
Sorry, I’m working today. A little bit in the rush. I’ll do it later on today, but it is the Cinco locomotive in the roster on this version of JMRI.Roster ID is one and locomotive is oneDarrin J. CalcuttSent from my iPhone

djcalcutt, 2024-08-17:
I’ll take a deeper look at it tonight when I get homeDarrin J. CalcuttSent from my iPhone

mstevetodd, 2024-08-17:
@djcalcutt 
I think I found what you are referring to. See screenshot.
[image]
I'm not sure when that was added, but it seems that only the JMRI Throttle was included, not webthrottle or WiThrottle clients.
The documentation [[here]](https://www.jmri.org/help/en/html/tools/throttle/RostersMediaPane.shtml) only refers to the JMRI Throttle UI:
`visible is that function button visible on throttle UI`
WiThrottle clients have always treated a blank label as an indication that the button should be hidden. I am not wure what assumptions are made by the WebThrottle client.
--SteveT

djcalcutt, 2024-08-18:
Yes thaT it it. Thank you very much! With web throttle i wanted to limit the general public to simple function. I just assumed that webthrottle was in compliance with the ‘visible’ check box. Can this be done?  Darrin J. CalcuttSent from my iPhone

image.png (view on web)

djcalcutt, 2024-08-18:
Aug 17, 2024 jmri part 1 | Synology Photosnas.cloudbusinessportal.caSo in testing… i cannot uncheck the visible check boxes when the lable is blank. Pictues 1. Just opened roster lables and medaPicture 2  uncheck visible and savePicture 3 reopen lables and media… check boxes are still checked. But…. If  enter something in as a lable, then the unchecked checkcox is saved See https://nas.cloudbusinessportal.ca/myMedia/mo/sharing/i9PQrwFLcBut still when i open  Withrottle or WebThrottle all functions show.  You are correct when you say  the throttle works correctlyAug 17, 2024 jmri 3 | Synology Photosnas.cloudbusinessportal.caDarrin J. CalcuttSent from my iPhone

djcalcutt, 2024-08-18:
In the mean time, can i manually edit the xml file?Darrin J. CalcuttSent from my iPhone

mstevetodd, 2024-08-20:
@djcalcutt, If you don't want a function key to show in a WebThrottle or WiThrottle client, just remove its label from the Roster entry.
Example from my demo layout attached.
[image]
[image]

djcalcutt, 2024-08-20:
Thank you very much! Darrin J. CalcuttSent from my iPhone

image.png (view on web)
image.png (view on web)

mstevetodd, 2024-08-23:
@djcalcutt Can this issue be closed?

djcalcutt, 2024-08-23:
Well, I guessBut I still don’t understand that when the visibility checkbox is unchecked,….Yes I just don’t understand how the item still show up in the web throttle when the checkboxes unchecked. Relying on an empty label doesn’t seem to be the correct way.Darrin J. CalcuttSent from my iPhone

closed 2024-11-25 by github-actions[bot]