JMRI/JMRI #13361: Include a means to override a Session Speed Profile updating Warrants track speeds during a session

NSW42220, 2024-08-20:
As far as I can tell a Session Speed Profile may/is created during a warrant run which may then amend the throttle/track speed ratio for the relevant RosterId/Address. This change is then in effect for that RosterId/Address during the session regardless of the Preferences>Warrants setting 'Speed Profile Merging to Roster'. 

Whilst the setting 'Speed Profile Merging to Roster' includes an option 'Don't Merge', it is applied upon exiting the session, not during the session. This means during a session, all Warrants for that RosterId/Address have their track speeds changed regardless of whether the operator desires or requires it, even if 'Don't Merge' was specified.

Currently the only work around is to delete all rows in the created Speed Profile after running the warrant. The original throttle/track speed ratio is then restored for that Roster Id/Address and thus track speed across the relevant warrant(s). However, it's only in effect if a new session speed profile is not created/applied.

A possible solution would be to include an option not to apply this speed profile during the session. This could be set within the Warrant create/edit panel (probably best) or as another option for 'Speed Profile Merging to Roster'.
Another possibility is an option to clear the Speed Profile at the end of the Warrant execution?

Thanks.

dsand47, 2024-08-26:
It looks like your analysis is correct.  The apparent goal is to fine tune the roster speed settings based on real world conditions.  For existing layouts using warrants, this was probably appropriate.

Adding a "speed setting" option to the warrant definition is probably the best approach.  The challenge is finding the location for such an option.

The warrant process involves the interactions between Warrant, OBlock, Engineer, Block Order and SpeedUtil.

I created a small test layout to try and simulate the speed setting issue.  Since I don't have any hardware, the simulation process does not create any speed setting changes.

If you are able to supply some supporting content I can take a look at finding a location for the option.  Attach a zip file to this issue with the following content:

- The layout configuration xml file.
- Any custom icons, etc.
- The roster/&lt;loco&gt;.xml file for the loco used for the test that has a speed profile.
- The signals/WarrantPreferences.xml file.
- The three messages.log files.
- A screenshot of the "View Speed Profile" after the warrant run which shows the differences.

Use the following steps to run the test:

1. Start PanelPro without loading a layout configuration file.
2. Select **Debug &rArr; Display / Edit Log Categories** from the main PanelPro menu.
3. Replace **Root Logging Level** with **jmri.jmrit.logix**.
4. Change DEBUG to TRACE
5. Click on **Set new Logging Level for this session**.
6. Manually load the layout configuration file.
7. Run the test warrant.
8. Get the speed profile screen shot.

The warrant tool has pretty good debugging output.  Hopefully the messages.log files will show what is needed.

NSW42220, 2024-08-26:
Thanks for your response. There certainly is a bit going on there.
Currently I'm just simulating warrant runs so there is no roster speed profile as such. BTW, I've also had some runs where a session speed profile wasn't created as the warrant track speeds didn't change.
I'll set aside some time this weekend and go through this methodically to gather up the relevant info.

NSW42220, 2024-09-01:
I ran the test as follows (simulated as I don't have a layout running):
NX Warrant created (point to point) then saved as TEST RUN SPEED ISSUE (screen shots)
Two test runs had no change and no speed profile created, the third did with warrant maximum speed changed and a speed profile created (screen shots)
Logs provided although I cleared the System Console before the third run.
N.B. I have a modified a version of BR2003 signaling system, but my customized masts are not in use as yet.
[WarrantSpeed_Issue.zip](https://github.com/user-attachments/files/16827324/WarrantSpeed_Issue.zip)

dsand47, 2024-09-02:
I have looked at your zip file contents.  It turns out that the Debug/Trace output is overwhelming.  

I have been adding targeted logging to the code that reduces the logging to a few lines per warrant block change.  I have also created a Jython program to handle the occupancy sensor changes when simulating a train being run by a warrant.  This makes it possible to have multiple runs with consistent timings.

When I ran the 9996 warrant I found that 350D was causing the creation of the speed profile change.  By reducing the script delay by one second, the speed profile change no longer occurred.  It turns out that short blocks are very sensitive to elapsed time across the block.

When PanelPro/DecoderPro is started, the roster.xml file is loaded into memory.  When a warrant is started, the "merge" list is checked. If it has a matching entry, its profile is assigned to the warrant.  This might include changes from a previous run.  If a merge entry is not present but there is a roster entry, the roster entry profile is copied to the merge list and assigned to the warrant.  If a merge entry  is not present and there is no roster entry, a temporary roster entry is created, an empty profile is added to the merge list and assigned to the warrant.  There is no separate session profile.  The session profile points to the corresponding profile in the merge list.  

The attached zip file contains a modified layout xml file and the Jython script.  These are moved to your user files location.  To use the script, start PanelPro and load the modified xml file.  
[image]
Click on 110D to make the block active and click on the right icon which invokes a LogixNG to load the Jython file.  The script waits for an internal sensor to become active.  The 9996 warrant has a sensor action to activate the left sensor.  When the script finishes successfully, the sensors are set inactive.

Note:  Errors and Java exceptions can occur.  There are a lot of threads involved and they can step on each other.  On one test, I had the simulated train run a signal displaying Stop.

[Simulation.zip](https://github.com/user-attachments/files/16832243/Simulation.zip)

NSW42220, 2024-09-03:
Many thanks Dave for your work on this. Ran the simulation and no issues.
From your description of merge, that explains why when I delete an entry the warrant reverts back to the original throttle/speed ratio.
Now whilst this avoided the issue happening and I understand why it applies the merge, I wouldn't mind the option to just let the warrant run 'as' is' without doing so i.e. not checking for a merge list.

P.S. Regarding the script, I can see how it may help with simulating block occupancy with other routes. Could I use it with my own layout file?
I'm not a coder what would I need to change in the script. It looks like I would need to amend the sensor list for the relevant route oblocks. What does the number represent e.g. 13 in sensorList.append(['BO 110D ARD HOME(N)', 13])

Much appreciated!

dsand47, 2024-09-03:
The number in each sensorList row is the number of seconds need to traverse the block.

The initial values are taken from a warrant session log.  The log was created by manually running the warrant and clicking the sensors when the "Overdue for arrival" message appears.  The elapsed time for block is from the log and then subtract at least 1 second from the time.  The 1 second is the delay before setting the previous block inactive.  

Subsequent runs when using the script verifies the timings.  You may get some very short overdue messages, those are ok.  If there are any longer ones, additional changes the times might be required.

The attached file is the session log that was used to build the script.

[WarrantSession.txt](https://github.com/user-attachments/files/16850701/WarrantSession.txt)

closed 2025-01-20 by github-actions[bot]