JMRI/JMRI #13366: Dispatcher issue

fert75, 2024-08-22:
Hi, When I try to run the train in Dispatcher it doesn't work. The train is still stopped and "waiting" I send my xlm file. Another question is if the blocks are correctly located at the double crossovers due to I will add blocks occupancy. I still have IR sensors.

Witch block occupancy hardware do you reccomend to work with JMRI and DCC ex Command station?

Thanks.

[TEST7_FT.txt](https://github.com/user-attachments/files/16713924/TEST7_FT.txt)

dsand47, 2024-08-22:
When Dispatcher does not work, go to "Help -> System Console".  Click on the "Copy to clipboard" button and paste into a reply.

dsand47, 2024-08-22:
The problem you are having is caused by a transit section action that sets the train speed before the train has been created.  

You have to set the action after the train is created.  Use "On Train Start" instead of "Sensor".

[image]

Your blocks are fine.  Note:  You have single crossovers, not double crossovers.

I don't know anything about DCC-EX.  You need to talk to DCC-EX people about occupancy.  Start at https://dcc-ex.com/#gsc.tab=0

fert75, 2024-08-23:
Thanks Dave! I send you the report of the System Console when I try to run a train in Dispatcher: 21:07:59,256 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]21:07:59,279 apps.util.Log4JUtil                   INFO  - This log is stored in file: C:\Users\FERTR\JMRI\log\session.log [main]21:07:59,280 apps.util.Log4JUtil                   INFO  - This log is appended to file: C:\Users\FERTR\JMRI\log\messages.log [main]21:07:59,313 apps.Apps                             INFO  - PanelPro version 5.8+Rbc21ce2ce7 starts under Java 22.0.1 on Windows 11 amd64 v10.0 at Thu Aug 22 21:07:59 ART 2024 [main]21:07:59,397 jmri.profile.ProfileManager           WARN  - Unable to set active profile. No profile with id My_JMRI_Railroad.3f016d01 could be found. [main]21:07:59,398 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3d14c7b9 [main]21:07:59,509 jmri.util.node.NodeIdentity           INFO  - Using d9a07b8c-69ba-4401-9bf7-c015fe6baf2f as the JMRI storage identity for profile id 3d14c7b9 [AWT-EventQueue-0]21:07:59,733 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "DCC++" [main]21:07:59,860 jmri.jmrix.dccpp.serial.DCCppAdapter  INFO  - Connecting DCC++ to COM3 USB Serial [main]21:07:59,864 jmri.jmrix.dccpp.serial.DCCppAdapter  INFO  - Port COM3 USB-SERIAL CH340 (COM3) opened at 115200 baud, sees DTR: true RTS: true DSR: false CTS: false DCD: false flow: NONE [main]21:07:59,985 mrix.dccpp.DCCppInitializationManager INFO  - DCC++ Initialization Complete [main]21:08:00,643 jmri.jmrit.roster.Roster              INFO  - Reading roster file with rootFromName(C:\Users\FERTR\JMRI\My_JMRI_Railroad.jmri\roster.xml) [main]21:08:01,013 jmri.util.startup.PerformFileModel    INFO  - Loading file C:\Users\FERTR\JMRI\My_JMRI_Railroad.jmri\TEST6_FT.txt.xml [main]21:08:03,876 jmri.implementation.VirtualSignalMast WARN  - attempting to set an aspect that has been disabled: Stop and Proceed on mast: SM TO4 B1 [AWT-EventQueue-0]21:08:03,877 implementation.DefaultSignalMastLogic ERROR - Error while creating Signal Logic [AWT-EventQueue-0]java.lang.IllegalArgumentException: attempting to set an aspect that has been disabled: Stop and Proceed on mast: SM TO4 B1               at jmri.implementation.VirtualSignalMast.setAspect(VirtualSignalMast.java:76) ~[jmri.jar:?]               at jmri.implementation.DefaultSignalMastLogic$DestinationMast.<init>(DefaultSignalMastLogic.java:1340) ~[jmri.jar:?]               at jmri.implementation.DefaultSignalMastLogic.setDestinationMast(DefaultSignalMastLogic.java:173) ~[jmri.jar:?]               at jmri.managers.configurexml.DefaultSignalMastLogicManagerXml.loadSignalMastLogic(DefaultSignalMastLogicManagerXml.java:226) ~[jmri.jar:?]               at jmri.managers.configurexml.DefaultSignalMastLogicManagerXml.load(DefaultSignalMastLogicManagerXml.java:200) ~[jmri.jar:?]               at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:632) ~[jmri.jar:?]               at jmri.configurexml.ConfigXmlManager.lambda$load$1(ConfigXmlManager.java:554) ~[jmri.jar:?]               at jmri.util.ThreadingUtil.lambda$runOnGUIwithReturn$1(ThreadingUtil.java:234) ~[jmri.jar:?]               at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:308) ~[?:?]               at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773) ~[?:?]               at java.awt.EventQueue$4.run(EventQueue.java:720) ~[?:?]               at java.awt.EventQueue$4.run(EventQueue.java:714) ~[?:?]               at java.security.AccessController.doPrivileged(AccessController.java:400) [?:?]               at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:87) [?:?]               at java.awt.EventQueue.dispatchEvent(EventQueue.java:742) [?:?]               at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) [?:?]               at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) [?:?]               at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) [?:?]               at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) [?:?]               at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) [?:?]               at java.awt.EventDispatchThread.run(EventDispatchThread.java:90) [?:?]21:08:03,898 jmri.implementation.VirtualSignalMast WARN  - attempting to set an aspect that has been disabled: Stop and Proceed on mast: SM TO3 B1 [AWT-EventQueue-0]21:08:03,898 implementation.DefaultSignalMastLogic ERROR - Error while creating Signal Logic [AWT-EventQueue-0]java.lang.IllegalArgumentException: attempting to set an aspect that has been disabled: Stop and Proceed on mast: SM TO3 B1               at jmri.implementation.VirtualSignalMast.setAspect(VirtualSignalMast.java:76) ~[jmri.jar:?]               at jmri.implementation.DefaultSignalMastLogic$DestinationMast.<init>(DefaultSignalMastLogic.java:1340) ~[jmri.jar:?]               at jmri.implementation.DefaultSignalMastLogic.setDestinationMast(DefaultSignalMastLogic.java:173) ~[jmri.jar:?]               at jmri.managers.configurexml.DefaultSignalMastLogicManagerXml.loadSignalMastLogic(DefaultSignalMastLogicManagerXml.java:226) ~[jmri.jar:?]               at jmri.managers.configurexml.DefaultSignalMastLogicManagerXml.load(DefaultSignalMastLogicManagerXml.java:200) ~[jmri.jar:?]               at j [... cut ...] r issue (Issue #13366) The problem you are having is caused by a transit section action that sets the train speed before the train has been created.You have to set the action after the train is created. Use "On Train Start" instead of "Sensor".Screenshot.2024-08-22.at.6.39.06.PM.png (view on web)Your blocks are fine. Note: You have single crossovers, not double crossovers.I don't know anything about DCC-EX. You need to talk to DCC-EX people about occupancy. Start at https://dcc-ex.com/#gsc.tab=0

dsand47, 2024-08-23:
That did not work well at all.

Next time just attach the "session.log" file located at "C:\Users\FERTR\JMRI\log".

Before you do that, fix the transit action to set the speed after "On Train Start".

You also need to clean up the signal errors.

1. Delete all of the signal mast logic.
2. Delete all of the C&amp;O-1980 signal masts.
3. Do a PanelPro store/stop/start/load.
4. Recreate the signal mast logic.

fert75, 2024-08-26:
Hi, I have an Dispatcher issue. When activate "Restart when done" the train when done it continues running in circle without following the actions that I setting in transit. No sound, no speed setting. Only do it in the first lap. I don't know what I'm doing wrong. Please help.

dsand47, 2024-08-26:
I have not used transit actions.  The JMRI user group has a much larger number of people actually using Dispatcher.

closed 2024-11-18 by github-actions[bot]