JMRI/JMRI #13370: Better way to display IP address(es), plus what servers are running on what ports

mstevetodd, 2024-08-23:
See https://groups.io/g/jmriusers/topic/108041262 for some background.

Currently, many users use the WiThrottle window to determine the hostname and IP that JMRI is running on, as well as the WiThrottle port. 

1) The WiThrottle UI uses the list of IPs from zeroconf for this display, so if anything prevents the zeroconf from working, it also affects this display. I believe this can be fixed by replacing the logic in WiThrottle.UserInterface.addIPAddessesToUI() with that in mailreport.ReportContext.addNetworkInfo(). Basically, get the ip list from networkInterface.getInetAddresses() instead of zeroconf.

2) A common request is to add a window to show that the WebServer is running, and what port it is running on. 

3) The Help, Context window shows all of this, but not in a user-friendly format.

Suggestion (for discussion): 
Add a new Help menu item "Network Servers"? which displays a list of the running servers and the port# for each, as well as a list of IP addresses and hostname for the current server.
--OR--
Add this information to the PanelPro main window in the area currently showing "Active Profile", etc.

danielb987, 2024-08-23:
> Add a new Help menu item "Network Servers"? which displays a list of the running servers and the port# for each, as well as a list of IP addresses and hostname for the current server.

I vote for this. It would be good if that window also has a startup action so that it can be opened when JMRI starts.

dsand47, 2024-08-23:
The IP information also needs to be available when running DecoderPro.  Some people run WiThrottle with DecoderPro.

Dave Sand

KenC57, 2024-08-23:
The Wi-Throttle window should show all connections it is listening on.
Users are used to looking at that to know what their phones should
connect to when the normal by name isn't working.

mstevetodd, 2024-08-24:
> The Wi-Throttle window should show all connections it is listening on.
> Users are used to looking at that to know what their phones should
> connect to when the normal by name isn't working.
> 
> 
This is not changing.

mstevetodd, 2024-08-24:
1) has been merged.

newHeiko, 2024-09-05:
Proposal: Add a "Zeroconf working" status? Otherwise people (=me ;) ) may wonder "I can see the IP address but my client won't autodetect the server".

Or do you think this will confuse users more than telling them "just type the IP into the client if it does not autodetect"?

Heiko

danielb987, 2024-09-05:
A suggestion is to add "autodiscover" to the message:

Zeroconf (autodiscover) is up.
Zeroconf (autodiscover) is down.

mstevetodd, 2024-09-05:
I could see stating "zeroconf (autodiscover) started" but there's no way to know "working". That depends on the network and the client(s).

danielb987, 2024-09-06:
If you don't get an exception, assume everything is working good.

closed 2024-11-25 by github-actions[bot]