JMRI/JMRI #13373: NCE PH5 AIU reading is broken

KenC57, 2024-08-24:
It seems that on a PH5 the AIU reply message is not being recognized by isSensorMessage()

I'm starting this which was seen by a user and by #13363 

My PH1 seems to work fine but it seems that the code at line 41 in NceSensorManager is never called. But the sensors update. Same thing, the line mInstance.handleSensorMessage(r); is never called and mInstance is alway null anywany. Something else is going on somewhere.

danielb987, 2024-08-24:
Thanks for looking at this. Once this is resolved, it would be very good if you could remove the bug pattern NP_UNWRITTEN_FIELD from `.spotbugs-check.xml` or ask me to do it.

See this change:
https://github.com/JMRI/JMRI/pull/13369/files#diff-3fab5c8a6b066dd701e88b1c995d278b9f5dfaaf937f872d3e46163ff2d9fb74

KenC57, 2024-08-25:
Right now I'm clearly getting 00 00 00 00 as my reply from a PH5. Other commands to it seem to reply just fine. Trying to find a good way to dig into it.

KenC57, 2024-08-27:
I'm likely to close this issue as it is pointing to the command station. More details soon.
Daniel, I think I added a line that will take care of the spotbugs issue.

KenC57, 2024-09-04:
I'm closing this as an issue since it traced back to a bug in the PH5 firmware. NCE will be updating their website with the fix and video to help users apply the update soon. I'm still going to push my other code changes that help track what connection the NCE is doing. If you have multiple connections, those different INFO messages didn't ID themselves well enough. I'm likely to do the same for a few other connections that are likely to have multiple connections some time later.

closed 2024-09-04 by KenC57